micropython/micropython #1844: Probably nobody cares, but flash is now filled for DEBUG=1 MICROPY_PY_WIZNET5K=1 MICROPY_PY_CC3K=1 build

dhylands, 2016-02-14:
I happened to have that particular combination as one of my travis builds, and that build now fills the flash (starting with commit c38809e26b71c539584462c2d02b6cd2647ade5b)

It looks like for PYBV10 that this is an artificial failure. The .ld file says that there is only 512K flash, but the STM32F405RGT6 (used on the PYBV10) actually has 1024K of flash.

pfalcon, 2016-02-14:
> DEBUG=1

Only now? I kinda got used that building with DEBUG=1 (-O0) will break expected sizes for many ports, and got used to hack around to do -Os -g.

dhylands, 2016-02-14:
The proverbial straw that broke the camels back :)

I run a bunch of extra builds:
https://github.com/dhylands/micropython/blob/travis-master/.travis.yml
which includes a bunch of DEBUG=1 builds, and it only just started failing.

And its not a real failure, since flash isn't really filled.

pfalcon, 2016-02-15:
Well, here's what I mean:

~~~
$ make BTARGET=application BTYPE=debug BOARD=LAUNCHXL
...
LINK build/LAUNCHXL/debug/application.axf
application.lds:102 cannot move location counter backwards (from 20041658 to 2003f400)
collect2: error: ld returned 1 exit status
make: *** [build/LAUNCHXL/debug/application.axf] Error 1
~~~

(The make command is the 1st appearing in cc3200/README.md,)

danicampora, 2016-02-15:
I noticed, will work on this today...

dpgeorge, 2017-06-26:
In 118173013f41a4f661bdb9baa88fb147e346ce75 I updated the F405 linker script to use the full 1MiB of flash.

closed 2017-06-26 by dpgeorge