micropython/micropython #1847: viper: ViperTypeError: can't do binary op between 'uint' and 'uint'

peterhinch, 2016-02-15:
The following produces the above error at compile time (only with uints - compiles and runs with int)
```python
@micropython.viper
def foo(arg1: uint, arg2: uint) -> uint:
    return arg1 + arg2
```
MicroPython v1.6-35-ge372e83 on 2016-02-11; PYBv1.0 with STM32F405RG

dpgeorge, 2016-02-15:
Unsigned (or signed/unsigned mixed) operations are not yet supported.

peterhinch, 2016-02-16:
For my doc, how can you use a uint? Is it only suitable for casting to a ptr (in which case you could just pass the ptr)?

dpgeorge, 2016-02-18:
> For my doc, how can you use a uint?

I think at the moment it's only useful for returning an integer, eg 0xffffffff.

dpgeorge, 2020-06-26:
Implemented in 41fa8b5482089bdd7fa5478fe24f32913b23967c

closed 2020-06-26 by dpgeorge