micropython/micropython #1848: should there be a "shutdown.py"

hoihu, 2016-02-15:
or similar? To shutdown/close initalized peripherals/connections gracefully after REPL ends?

There was a discussion in #733 with the temporary outcome that the timer callbacks are deinit. However it seems that also #1796 is affected by this (USB VCP handling of CTRLC). 

Anyway, just thought a "shutdown.py" might be a solution for that. If present it would be executed if CTRLD was pressed. Would be kind of an analogy to "boot.py"...

hoihu, 2016-03-07:
based on the feedback probably not ;)

I'm closing this now.

dpgeorge, 2016-03-07:
The REPL is really only used for debugging and prototyping, so providing a shutdown mechanism isn't really necessary.  Shutdown should be handled by your "main.py" when it's about to finish/reset.  "boot.py" can also handle shutdown by working out what the reset cause was.

hoihu, 2016-03-23:
I use CTRLDs to import the changes that i've uploaded to the board... Correct me but  main.py has no means to detect CTRLD (in Python), so it is not possible to handle that there? 

boot.py seems like the better option, but is there a way to evaluate the reset cause (e.g. ctrld?).

hoihu, 2016-09-29:
since the machine module supports now the read out of the reset cause, I think this PR can be closed.

To summarize:
- Repl is not intended to handle shutdown mechanism. 
- Use main.py for shutdown procedures or read out reset cause at startup to init/deinit peripherals

closed 2016-09-29 by hoihu