micropython/micropython #1852: objrange: No check for zero step in range() function.

hosaka, 2016-02-17:
While attempting to follow previously raised #1711 and #1481.
```python
>>> for i in range(0, 10, 0):
...     print(i)
...  
```
Will cause an infinite loop, printing 1 until ctrl-c is sent. While cpython prints:
```
ValueError: range() arg 3 must not be zero
```
There is actually a TODO note in [objrange.c#L107](https://github.com/micropython/micropython/blob/master/py/objrange.c#L107). Since the patch is tiny, I will probably include it while working on negative and zero step slicing issues, unless there will be other recommendations.

hosaka, 2016-02-17:
So I didn't know that `for x in range():` does not create an iterator, from what I can see. How and where in code is the optimised iteration handled in a for-range case? I looked at for_range.py, but no obvious hints.

dpgeorge, 2016-02-17:
> How and where in code is the optimised iteration handled in a for-range case?

https://github.com/micropython/micropython/blob/master/py/compile.c#L1331

hosaka, 2016-02-17:
Thank you Damien. I will probably open a separate PR for this.

pfalcon, 2016-02-17:
> Will cause an infinite loop, printing 1 until ctrl-c is sent. 

Cool, you asked it for infinite loop, it gave it to you. Cool?

> While cpython prints:

Then micropython is more powerful than cpython - it allows to do more instead of throwing out excuses ;-).

> other recommendations.

Unless this solves some problem (like, last issue to make Django run on uPy), I'd leave it. There's no crash, etc. We still have cases where uPy crashes - they should be fixed.

hosaka, 2016-02-17:
Yes, I understand what it gives me, and it's doing nothing wrong! :) It doesn't crash, but it puts you in a situation, that is difficult to escape, unless you're in REPL. I dare suggest another option, in compile.c#L1331: 
```c
    if (MP_PARSE_NODE_LEAF_SMALL_INT(pn_step) >= 0) {
        EMIT_ARG(binary_op, MP_BINARY_OP_LESS);
    } else {
        EMIT_ARG(binary_op, MP_BINARY_OP_MORE);
    }
```
Change `>=` to simply `>`. At least the for loop will do nothing instead of running infinitely. If not, I'll leave this in peace and try work on the negative step slicing.

hosaka, 2016-02-17:
Similar as [compile.c#L1990](https://github.com/micropython/micropython/blob/master/py/compile.c#L1990) has:
```c
compile_syntax_error(comp, MP_PARSE_NODE_NULL, "super() call cannot find self"); // really a TypeError
return;
```
The above range(0,10,0) would be a ValueError, rather than syntax, it's best not to complicate things even further.

hosaka, 2016-02-26:
Closing this as non-issue.

closed 2016-02-26 by hosaka