micropython/micropython #1858: Access Frozen Module from SD Card

leifzars, 2016-02-26:
I am considering developing a large application in microPython, and am concerned with the limited resources I have on my setup. 
Can I precompile and freeze modules and dynamically load them from the SD card? Basically page in and out of RAM modules as they are used? If so how good of an idea is this, or is it not?

I hate to start developing a project on micropython and find out that I can't store and or access all my code do to the limited amount of flash and ram I have.

leifzars, 2016-03-06:
Does this question not make sense ?

dhylands, 2016-03-06:
Frozen modules exist in flash, and currently, then only way to get them there is to rebuild the firmware.

You can precompile your python source to create bytecode (foo.mpy instead off foo.py files) and store the mpy files on the sdcard.

You should be able to unload a previously imported module by doing something like:
```
import sys
import gc
del sys.module['module_name']
gc.collect()
```
However if there are references to a function in your module (say a callback) then that won't get reclaimed.

dpgeorge, 2016-03-07:
@leifzars what you ask for is not possible (yet).  There is no paging mechanism.  It would be difficult because when you import a module in can execute arbitrary code and create arbitrary objects, which can then be updated in arbitrary ways during use of that module.  So to "page it out" would require a generic paging mechanism/solution to page arbitrary stuff out of the heap.  Essentially you'd implement a memory subsystem whereby the interal RAM was a cache for storage of the heap on an external SD card.

As @dhylands suggests, you can store precompiled .mpy files on the SD card, but when you import them they use up RAM to copy across the bytecode.

Using proper frozen bytecode (see #1811) would allow you to put all your Python code in flash.  And there's a lot of flash, you should be able to get 10k+ lines of Python in there.  Then the only limit that RAM gives is how many objects you can create at runtime.

pfalcon, 2016-03-23:
Frozen modules are by definition is what's included inside an interpreter binary.

closed 2016-03-23 by pfalcon