micropython/micropython #1862: ussl.wrap_socket silently accepts invalid certificates

ncoghlan, 2016-02-28:
Investigating https://github.com/micropython/micropython-lib/issues/69, I found the current SSL/TLS socket creation code at https://github.com/micropython/micropython/blob/d19e4f0ba4df487b4ebd36b5fe6a16e68c0afe77/extmod/modussl.c#L49

If I'm reading that correctly:

1. Wrapping a socket without providing any certificate verification details results in no verification being performed;
2. Even if verification details are provided, they're still ignored

This makes the documentation at http://docs.micropython.org/en/latest/library/ussl.html#ussl.ssl.wrap_socket thoroughly misleading, as even if the additional arguments are passed in, they won't be processed.

I realise actually implementing this will require a significant amount of work, so my request at this point would be for passing in unsupported arguments to result in a hard failure, rather than silently appearing to succeed without actually providing the claimed security guarantees.

ncoghlan, 2016-02-28:
For background on the changes that were made to CPython's default behaviour in this area: https://www.python.org/dev/peps/pep-0476/

For MicroPython, the idea of relying on the system certificate stores isn't valid (since those interfaces don't exist on embedded systems), hence the idea in https://github.com/micropython/micropython-lib/issues/69#issuecomment-189804139 of expecting either the Mozilla cert bundle or dedicated security certificates to be available when creating HTTPS connections, and refusing to create them otherwise.

danicampora, 2016-02-28:
> the idea in micropython/micropython-lib#69 (comment)

I should point out that the docs in
http://docs.micropython.org/en/latest/library/ussl.html#ussl.ssl.wrap_socket
are valid but specific to the WiPy. We should make that clear and also
separate what applies to the other ports.

sigmavirus24, 2016-02-28:
See also https://cve.mitre.org/cgi-bin/cvename.cgi?name=2014-9365. I would argue a separate CVE should be assigned for this behaviour in micropython.

pfalcon, 2016-05-22:
https://github.com/micropython/micropython-lib/issues/69#issuecomment-220852192 explains that initial user story behind ussl module was "make over-zealous HTTPS services work as HTTP". Stakeholders of other user stories are welcome to share them here, gather together, and find a solution to this issue.

pfalcon, 2016-05-22:
https://github.com/micropython/micropython/commit/bca4c9e465efe1160fcdf0f7467f47d9e6175879 adds  disclosure of cert validation behavior to the official docs. Note that this is first time docs for axTLS-based ussl module are published, 2 weeks after first release of the port where it was enabled (esp8266 port). unix port still doesn't have it enabled (something to do soon finally).

pfalcon, 2016-05-22:
wipy is not susceptible to the issue as described (it has support for cert validation; the implementation is not open-source from the upstream vendor (TI), so cannot be audited for omissions, bugs, or backdoors).