micropython/micropython #1863: Error 5 when reading after seek

robert-hh, 2016-03-01:
I have this sample sketch which will create an error 5:
```
b = bytearray(1000)
with open("sample.dat", "rb") as f:
    f.seek(511) 
    n = f.readinto(b)
print("Done, bytes read: ", n)
```
sample.dat is an arbitrary file of sufficient length. If I write f.seek(512), all is well.

MicroPython v1.6-92-gac23662 on 2016-02-25; PYBv1.0 with STM32F405RG

dpgeorge, 2016-03-01:
I can't confirm this.  My set up is: MicroPython v1.6-107-g4f72aa8 on 2016-03-01; PYBv1.0 with STM32F405RG, and the file that's being read is 2000 bytes long.  It works as expected (ie can read 1000 bytes successfully) with a seek of 511 and 512.  Do you use internal flash or SD card?

peterhinch, 2016-03-02:
I have replicated this, on SD but not flash
```
>>> 
PYB: sync filesystems
PYB: soft reboot
MicroPython v1.6-99-g42bc25b-dirty on 2016-03-01; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== b = bytearray(1000)
=== with open("/sd/timesroman45x46", "rb") as f:
===     f.seek(511) 
===     n = f.readinto(b)
=== print("Done, bytes read: ", n)
=== 
511
Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
  File "<stdin>", line 4, in <module>
OSError: 5
>>> 
```

robert-hh, 2016-03-02:
I'm using SD Card. And it does not matter, whether it's a seek or another shorter read. Same with version 107. But with flash it works! Funny. I had never seen a difference before. I tried another brand new freshly formatted SD card. Same error! It also depends on the size & sequence of reads.
```
This one fails:

with open("sample.dat", "rb") as f:
    b = bytearray(1098)    
    n = f.readinto(b)
    b = bytearray(960)
    n = f.readinto(b)
print("Done, bytes read: ", n)
---------------------------

These two pass

with open("sample.dat", "rb") as f:
    b = bytearray(960)    
    n = f.readinto(b)
    b = bytearray(1098)
    n = f.readinto(b)
print("Done, bytes read: ", n)
-----------------------------
with open("sample.dat", "rb") as f:
    b = bytearray(1000)    
    n = f.readinto(b)
    b = bytearray(800)
    n = f.readinto(b)
print("Done, bytes read: ", n)
```
MicroPython v1.6-109-g6bb9d3e on 2016-03-02; PYBv1.0 with STM32F405RG

peterhinch, 2016-03-02:
A thought. I routinely do random read access of binary files on an SD card using the official sdcard driver. So I guess the fault is restricted to /sd.

robert-hh, 2016-03-02:
Another thought. I'll test if the fails happens when the first read/seek is NOT a multiple of a specific value, like 4 or 16. The passed variants have that property, the failed one not. But I have to wait until the evening.
P.S.: Hi Peter, I see that you were up early, like 6 AM

robert-hh, 2016-03-02:
So I have a result. The code works if the seek (or the number of bytes in the first read) is a multiple of 4. Maybe that observation helps.

dpgeorge, 2016-03-03:
I confirm this bug.  The issue is that the built-in SD card driver requires the buffer for read/write to be aligned on a 4-byte boundary.  This is not the case if you are at a non-aligned location in the file and you ask to read more than 1 sector worth.  Simplest test that fails:
```python
f = open('test', 'rb')
f.seek(511)
f.read(513)
```

robert-hh, 2016-03-03:
Hello Damien, thanks for the reply. That is what I considered for trying while sitting in my day-time office - to split a read crossing the sector boundary into a shorter read up to a sector boundary first and then the remainder. That might help as a work-around, at least, but is truly inefficient.

dpgeorge, 2016-05-13:
Fixed; see above commit.

closed 2016-05-13 by dpgeorge

robert-hh, 2016-05-13:
Hello Damien,

thank you. It works in my test cases.

Best Regards, Robert