micropython/micropython #1864: esp8266 compilation breaks because of `rcons`

drasko, 2016-03-05:
Here is the log
```
...
CC ../py/../extmod/modubinascii.c
CC ../py/../extmod/modurandom.c
CC ../py/../extmod/fsusermount.c
CC strtoll.c
CC esp_mphal.c
CC main.c
CC gccollect.c
CC uart.c
CC modpyb.c
CC modpybpin.c
CC modpybrtc.c
CC modpybadc.c
In file included from /uiota/esp-open-sdk/esp_iot_sdk_v1.5.0/include/ets_sys.h:12:0,
                 from uart.c:12:
uart.c: In function 'uart_config':
/uiota/esp-open-sdk/esp_iot_sdk_v1.5.0/include/eagle_soc.h:249:38: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
                                      &  (~(PERIPHS_IO_MUX_FUNC<<PERIPHS_IO_MUX_FUNC_S))  \
                                      ^
/uiota/esp-open-sdk/esp_iot_sdk_v1.5.0/include/eagle_soc.h:50:98: note: in definition of macro 'WRITE_PERI_REG'
 #define WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val)
                                                                                                  ^
uart.c:47:9: note: in expansion of macro 'PIN_FUNC_SELECT'
         PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO2_U, FUNC_U1TXD_BK);
         ^
/uiota/esp-open-sdk/esp_iot_sdk_v1.5.0/include/eagle_soc.h:249:38: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
                                      &  (~(PERIPHS_IO_MUX_FUNC<<PERIPHS_IO_MUX_FUNC_S))  \
                                      ^
/uiota/esp-open-sdk/esp_iot_sdk_v1.5.0/include/eagle_soc.h:50:98: note: in definition of macro 'WRITE_PERI_REG'
 #define WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val)
                                                                                                  ^
uart.c:51:9: note: in expansion of macro 'PIN_FUNC_SELECT'
         PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD);
         ^
/uiota/esp-open-sdk/esp_iot_sdk_v1.5.0/include/eagle_soc.h:249:38: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
                                      &  (~(PERIPHS_IO_MUX_FUNC<<PERIPHS_IO_MUX_FUNC_S))  \
                                      ^
/uiota/esp-open-sdk/esp_iot_sdk_v1.5.0/include/eagle_soc.h:50:98: note: in definition of macro 'WRITE_PERI_REG'
 #define WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val)
                                                                                                  ^
uart.c:52:9: note: in expansion of macro 'PIN_FUNC_SELECT'
         PIN_FUNC_SELECT(PERIPHS_IO_MUX_MTDO_U, FUNC_U0RTS);
         ^
CC modnetwork.c
CC modesp.c
CC modutime.c
CC moduos.c
CC build/frozen.c
CC utils.c
AS gchelper.s
In file included from /uiota/esp-open-sdk/esp_iot_sdk_v1.5.0/include/ets_sys.h:12:0,
                 from /uiota/esp-open-sdk/esp_iot_sdk_v1.5.0/include/os_type.h:10,
                 from /uiota/esp-open-sdk/esp_iot_sdk_v1.5.0/include/user_interface.h:9,
                 from modpybpin.c:32:
modpybpin.c: In function 'pyb_pin_obj_init_helper':
/uiota/esp-open-sdk/esp_iot_sdk_v1.5.0/include/eagle_soc.h:249:38: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
                                      &  (~(PERIPHS_IO_MUX_FUNC<<PERIPHS_IO_MUX_FUNC_S))  \
                                      ^
/uiota/esp-open-sdk/esp_iot_sdk_v1.5.0/include/eagle_soc.h:50:98: note: in definition of macro 'WRITE_PERI_REG'
 #define WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val)
                                                                                                  ^
modpybpin.c:94:5: note: in expansion of macro 'PIN_FUNC_SELECT'
     PIN_FUNC_SELECT(self->periph, self->func);
     ^
CC ../stmhal/pybstdio.c
CC ../lib/libc/string0.c
CC ../lib/netutils/netutils.c
CC ../lib/mp-readline/readline.c
CC ../lib/timeutils/timeutils.c
CC ../lib/utils/pyexec.c
CC ../lib/utils/printf.c
LINK build/firmware.elf
/uiota/esp-open-sdk/esp_iot_sdk_v1.5.0/lib/libwpa.a(aes-internal.o):(.irom0.text+0x0): undefined reference to `rcons'
Makefile:127: recipe for target 'build/firmware.elf' failed
make: *** [build/firmware.elf] Error 1
```

Can be realted to this issue: https://github.com/sticilface/ESPmanager/issues/3

Any ideas?

drasko, 2016-03-05:
Mentioned here also: https://github.com/platformio/platformio-pkg-ldscripts/issues/1

dpgeorge, 2016-03-07:
Try building with ESP SDK v1.5.2.

pfalcon, 2016-03-23:
Please see http://forum.micropython.org/viewtopic.php?f=16&t=1689 for further instructions.

closed 2016-03-23 by pfalcon