micropython/micropython #1866: Do the micropython can control the AT89C51?

wangaicc, 2016-03-05:
Do the micro python can control the AT89C51?

gauravsharma013, 2016-03-05:
It can work, but no one developing on it yet. Why don't you initiate :)

dpgeorge, 2016-03-07:
The AT89C51 has 4k flash and 128 bytes of RAM.  There is no way MicroPython can fit on there with such little memory.

The AT89C51 can access external memory via port 0 and port 2 (up to 64k) so you might be able to run a subset of MicroPython this way, but I don't think it's a useful thing to do. 

MicroPython running on another MCU can *control* an external AT89C51, if that's what you are asking.

closed 2016-03-07 by dpgeorge

wangaicc, 2016-03-22:
thank you.
I just want to know about micro python.
I am just a high school teacher.
I thought could use Python to write programs that would be like using C language to control AT89C51.
if the micro python can work.
then I can to be able to give the child a little with Python courses.
thank you