micropython/micropython #1877: micropython.mem_info() possible memory lead

noelhenson, 2016-03-07:
Every time micropython.mem_info() is run, it seems that 64 bytes of available memory is lost.
MicroPython v1.6-106-g2740dd8 on 2016-02-29; PYBv1.0 with STM32F405RG

Repro:
>>> micropython.mem_info()
stack: 424 out of 15360
GC: total: 102400, used: 8768, free: 93632
 No. of 1-blocks: 168, 2-blocks: 70, max blk sz: 35
>>> micropython.mem_info()
stack: 424 out of 15360
GC: total: 102400, used: 8832, free: 93568
 No. of 1-blocks: 170, 2-blocks: 71, max blk sz: 35
>>> micropython.mem_info()
stack: 424 out of 15360
GC: total: 102400, used: 8896, free: 93504
 No. of 1-blocks: 172, 2-blocks: 72, max blk sz: 35
>>> micropython.mem_info()
stack: 424 out of 15360
GC: total: 102400, used: 8960, free: 93440
 No. of 1-blocks: 174, 2-blocks: 73, max blk sz: 35
>>> micropython.mem_info()
stack: 424 out of 15360
GC: total: 102400, used: 9024, free: 93376
 No. of 1-blocks: 176, 2-blocks: 74, max blk sz: 35
>>>

stinos, 2016-03-07:
Adding calls to ```gc.collect()``` in between will probably give you different, less 'leaky', results

dpgeorge, 2016-03-07:
@stinos is correct: the used/free info reported by mem_info() does not take into account possible garbage.  Running the command each time requires compiling to bytecode and executing, which uses memory.  This will be reclaimed when a garbage collection is run.

closed 2016-03-07 by dpgeorge

noelhenson, 2016-03-07:
Damien,

Thanks for the personal response. As soon as I read the response from 
stinos, I realized what was going on. I'm new to micropython and still 
figuring things out.

Noel