micropython/micropython #1881: esp8266 compilation fails - missing lwip/ip_addr.h

ryannathans, 2016-03-08:
using the latest pulled from https://github.com/pfalcon/esp-open-sdk

```
$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/qstrdefs.generated.h
Generating build/genhdr/mpversion.h
mkdir -p build/build/
mkdir -p build/lib/libc/
mkdir -p build/lib/libm/
mkdir -p build/lib/mp-readline/
mkdir -p build/lib/netutils/
mkdir -p build/lib/timeutils/
mkdir -p build/lib/utils/
mkdir -p build/py/
mkdir -p build/py/../extmod/
mkdir -p build/stmhal/
CC ../py/mpstate.c
CC ../py/nlrx86.S
CC ../py/nlrx64.S
CC ../py/nlrthumb.S
CC ../py/nlrxtensa.S
CC ../py/nlrsetjmp.c
CC ../py/malloc.c
CC ../py/gc.c
CC ../py/qstr.c
CC ../py/vstr.c
CC ../py/mpprint.c
CC ../py/unicode.c
CC ../py/mpz.c
CC ../py/lexer.c
CC ../py/lexerstr.c
CC ../py/lexerunix.c
CC ../py/parse.c
CC ../py/scope.c
CC ../py/compile.c
CC ../py/emitcommon.c
CC ../py/emitbc.c
CC ../py/asmx64.c
CC ../py/emitnative.c
CC ../py/asmx86.c
CC ../py/emitnative.c
CC ../py/asmthumb.c
CC ../py/emitnative.c
CC ../py/emitinlinethumb.c
CC ../py/asmarm.c
CC ../py/emitnative.c
CC ../py/formatfloat.c
CC ../py/parsenumbase.c
CC ../py/parsenum.c
CC ../py/emitglue.c
CC ../py/runtime.c
CC ../py/nativeglue.c
CC ../py/stackctrl.c
CC ../py/argcheck.c
CC ../py/warning.c
CC ../py/map.c
CC ../py/obj.c
CC ../py/objarray.c
CC ../py/objattrtuple.c
CC ../py/objbool.c
CC ../py/objboundmeth.c
CC ../py/objcell.c
CC ../py/objclosure.c
CC ../py/objcomplex.c
CC ../py/objdict.c
CC ../py/objenumerate.c
CC ../py/objexcept.c
CC ../py/objfilter.c
CC ../py/objfloat.c
CC ../py/objfun.c
CC ../py/objgenerator.c
CC ../py/objgetitemiter.c
CC ../py/objint.c
CC ../py/objint_longlong.c
CC ../py/objint_mpz.c
CC ../py/objlist.c
CC ../py/objmap.c
CC ../py/objmodule.c
CC ../py/objobject.c
CC ../py/objpolyiter.c
CC ../py/objproperty.c
CC ../py/objnone.c
CC ../py/objnamedtuple.c
CC ../py/objrange.c
CC ../py/objreversed.c
CC ../py/objset.c
CC ../py/objsingleton.c
CC ../py/objslice.c
CC ../py/objstr.c
CC ../py/objstrunicode.c
CC ../py/objstringio.c
CC ../py/objtuple.c
CC ../py/objtype.c
CC ../py/objzip.c
CC ../py/opmethods.c
CC ../py/sequence.c
CC ../py/stream.c
CC ../py/binary.c
CC ../py/builtinimport.c
CC ../py/builtinevex.c
CC ../py/modarray.c
CC ../py/modbuiltins.c
CC ../py/modcollections.c
CC ../py/modgc.c
CC ../py/modio.c
CC ../py/modmath.c
CC ../py/modcmath.c
CC ../py/modmicropython.c
CC ../py/modstruct.c
CC ../py/modsys.c
CC ../py/vm.c
CC ../py/bc.c
CC ../py/showbc.c
CC ../py/repl.c
CC ../py/smallint.c
CC ../py/frozenmod.c
CC ../py/../extmod/moductypes.c
CC ../py/../extmod/modujson.c
CC ../py/../extmod/modure.c
CC ../py/../extmod/moduzlib.c
CC ../py/../extmod/moduheapq.c
CC ../py/../extmod/moduhashlib.c
CC ../py/../extmod/modubinascii.c
CC ../py/../extmod/machine_mem.c
CC ../py/../extmod/modussl.c
CC ../py/../extmod/modurandom.c
CC ../py/../extmod/fsusermount.c
CC ../py/../extmod/vfs_fat.c
CC ../py/../extmod/vfs_fat_ffconf.c
CC ../py/../extmod/vfs_fat_diskio.c
CC ../py/../extmod/vfs_fat_file.c
CC ../py/../extmod/vfs_fat_lexer.c
CC ../py/../extmod/vfs_fat_misc.c
CC ../py/../extmod/moduos_dupterm.c
CC strtoll.c
CC main.c
In file included from main.c:39:0:
/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/user_interface.h:11:26: fatal error: lwip/ip_addr.h: No such file or directory
 #include "lwip/ip_addr.h"
                          ^
compilation terminated.
../py/mkrules.mk:47: recipe for target 'build/main.o' failed
make: *** [build/main.o] Error 1

```

ryannathans, 2016-03-08:
i'm actually thinking micropython isn't to blame here...

pfalcon, 2016-03-08:
esp8266 port is in flux coming weeks. Getting tickets like this may only slow down us... (Aka, yes, build is broken, will be fixed, then broken again - many times.)

ryannathans, 2016-03-08:
@pfalcon alright, I understand. Close this issue if need be

Are there plans in the works to have a "nightly" build?

closed 2016-03-08 by ryannathans