micropython/micropython #1883: Allow to run .mpy files by specifying them on command line.

prusnak, 2016-03-08:
One can build a .mpy files using mpy-cross compiler. These can be further used via command-line using the following:

```
mpy-cross foo.py # creates foo.mpy
micropython -c 'import foo' # runs foo.mpy
```

However this does not work:
```
micropython foo.mpy
```

It would be benificial if this was possible (maybe via some command line option), so micropython could skip parser phase completely. Maybe, we could even save space, be compiling a special version of micropython without the parser.

prusnak, 2016-03-15:
Another effect would be that bytecode fuzzing using AFL (in PR https://github.com/micropython/micropython/pull/1884) could be much faster if parser is completely skipped.

dpgeorge, 2018-12-04:
It is possible to run .mpy files directly from the command line using the "-m" switch:
```
$ micropython -m foo
```
This is not 100% ideal because behind the scenes it does an "import foo", which could import some other foo module, but it does work and skips the parser/compiler completely.

prusnak, 2018-12-04:
Thanks, I think we can close the issue.

closed 2018-12-04 by prusnak