micropython/micropython #1885: esp8266 build error

skylin008, 2016-03-09:
when build the esp8266 firmware,the error message shows follow:

fff@cc-b1c25b7d7454 /micropython/esp8266
$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosit
y.
CC main.c
In file included from D:/msys/1.0/opt/include/ets_sys.h:11:0,
                 from D:/msys/1.0/opt/include/os_type.h:10,
                 from D:/msys/1.0/opt/include/user_interface.h:9,
                 from main.c:39:
D:/msys/1.0/opt/include/c_types.h:15:29: error: conflicting types for 'uint32_t'

 typedef unsigned long       uint32_t;
                             ^
In file included from d:\sysgcc\esp8266\lib\gcc\xtensa-lx106-elf\5.2.0\include\s
tdint.h:9:0,
                 from ./mpconfigport.h:1,
                 from ../py/mpconfig.h:45,
                 from ../py/nlr.h:36,
                 from main.c:30:
d:\sysgcc\esp8266\xtensa-lx106-elf\sys-include\stdint.h:81:22: note: previous de
claration of 'uint32_t' was here
 typedef unsigned int uint32_t;
                      ^
In file included from D:/msys/1.0/opt/include/ets_sys.h:11:0,
                 from D:/msys/1.0/opt/include/os_type.h:10,
                 from D:/msys/1.0/opt/include/user_interface.h:9,
                 from main.c:39:
D:/msys/1.0/opt/include/c_types.h:17:29: error: conflicting types for 'int32_t'
 typedef signed long         int32_t;
                             ^
In file included from d:\sysgcc\esp8266\lib\gcc\xtensa-lx106-elf\5.2.0\include\s
tdint.h:9:0,
                 from ./mpconfigport.h:1,
                 from ../py/mpconfig.h:45,
                 from ../py/nlr.h:36,
                 from main.c:30:
d:\sysgcc\esp8266\xtensa-lx106-elf\sys-include\stdint.h:80:20: note: previous de
claration of 'int32_t' was here
 typedef signed int int32_t;
                    ^
In file included from ../py/misc.h:33:0,
                 from ../py/qstr.h:30,
                 from ../py/lexer.h:32,
                 from ../py/compile.h:29,
                 from main.c:31:
D:/msys/1.0/opt/include/c_types.h:85:25: error: two or more data types in declar
ation specifiers
 typedef unsigned char   bool;
                         ^
In file included from D:/msys/1.0/opt/include/ets_sys.h:11:0,
                 from D:/msys/1.0/opt/include/os_type.h:10,
                 from D:/msys/1.0/opt/include/user_interface.h:9,
                 from main.c:39:
D:/msys/1.0/opt/include/c_types.h:85:1: error: useless type name in empty declar
ation [-Werror]
 typedef unsigned char   bool;
 ^
D:/msys/1.0/opt/include/c_types.h:87:0: error: "true" redefined [-Werror]
 #define true            (1)
 ^
In file included from ../py/misc.h:33:0,
                 from ../py/qstr.h:30,
                 from ../py/lexer.h:32,
                 from ../py/compile.h:29,
                 from main.c:31:
d:\sysgcc\esp8266\lib\gcc\xtensa-lx106-elf\5.2.0\include\stdbool.h:34:0: note: t
his is the location of the previous definition
 #define true 1
 ^
In file included from D:/msys/1.0/opt/include/ets_sys.h:11:0,
                 from D:/msys/1.0/opt/include/os_type.h:10,
                 from D:/msys/1.0/opt/include/user_interface.h:9,
                 from main.c:39:
D:/msys/1.0/opt/include/c_types.h:88:0: error: "false" redefined [-Werror]
 #define false           (0)
 ^
In file included from ../py/misc.h:33:0,
                 from ../py/qstr.h:30,
                 from ../py/lexer.h:32,
                 from ../py/compile.h:29,
                 from main.c:31:
d:\sysgcc\esp8266\lib\gcc\xtensa-lx106-elf\5.2.0\include\stdbool.h:35:0: note: t
his is the location of the previous definition
 #define false 0
 ^
In file included from main.c:39:0:
D:/msys/1.0/opt/include/user_interface.h:11:26: fatal error: lwip/ip_addr.h: No
such file or directory
cc1.exe: all warnings being treated as errors
compilation terminated.
make: *** [build/main.o] Error 1

How to solve this problem.Thanks advance!

ryannathans, 2016-03-09:
Please see: #1881

dpgeorge, 2016-03-09:
Did you upgrade to the latest esp-open-sdk?  Try deleting it and cloning/build the SDK again.

skylin008, 2016-03-14:
Thanks @dpgeorge .I had upadte the latest version,but the other issues shows,follow the error message:/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -lmain
/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -ljson
/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -lssl
/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -llwip_open
/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -lpp
/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -lnet80211
/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -lwpa
/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -lphy
/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -lnet80211
make:***[build/firmware.elf]] Error1.

How to solve it ,TKS advance!

pfalcon, 2016-03-23:
Please see http://forum.micropython.org/viewtopic.php?f=16&t=1689 for further instructions.

closed 2016-03-23 by pfalcon