micropython/micropython #1886: Question about Timer

iitzex, 2016-03-09:
```python
from time import sleep_us, sleep_ms
from machine import Timer

def count():
    for i in range(5):
        print(i)
        sleep_ms(500)

tim = Timer(0)
tim.init(period=5000, mode=Timer.PERIODIC, callback=lambda t:count())
```
Why system got reset if I using sleep or delay?

dhylands, 2016-03-09:
Set the timer to run every 500 msec, and have it count to 10, but only print something if the counter is less than 5.

Here's an example for the pyboard which does something similar:
https://github.com/dhylands/upy-examples/blob/master/heartbeat_irq.py

iitzex, 2016-03-09:
Thank you for the example, and I try to know more.
In just normal case, what will cause wdt reset?
15ms or 500ms

dhylands, 2016-03-09:
I'm not sure. Sleeping (for any amount of time) from within an ISR is generally a bad idea because often times the sleep function require other ISR functions which may be disabled because you're within an ISR.

iitzex, 2016-03-09:
I try to put some delay in the blinky example, but it seems cause wdt reset. I want to know what interval of delay will make it reset. Some examples suggest using timer. That's why I write the code above to figure it out, but I'm still not know detail reason of wdt reset. Thanks for the suggestion.

closed 2016-03-25 by iitzex