micropython/micropython #1892: Standalone binary?

VanushVaswani, 2016-03-11:
Is it possible to create a standalone executable of a Python script using MicroPython as a "backend", similar to the Nuitka project?

dpgeorge, 2016-03-11:
No, it's not currently possible to do this, although we are very close.  Please continue the discussion in #1295.

closed 2016-03-11 by dpgeorge

VanushVaswani, 2016-03-11:
Thanks. 

I actually think frozen bytecode in #1811 will achieve what I'm after. Since my platform has no REPL or file system, is there a way to run the static bytecode, i.e `run_bytecode`?

dpgeorge, 2016-03-11:
> Since my platform has no REPL or file system, is there a way to run the static bytecode, i.e run_bytecode?

Yes, `pyexec_frozen_module()`.  That PR comes with an example that works on pyboard.