micropython/micropython #1894: make: *** [build/ets_alt_task.o] Error 1

bisguzar, 2016-03-13:
hello everyone, i have a problem. when i do compile micropython for esp8266,(nodemcu) on ubuntu i see an error.

`In file included from ets_alt_task.c:5:0:
/home/vagrant/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/esp_sdk_ver.h:1:1: error: expected identifier or '(' before '-' token
 -e #undef ESP_SDK_VERSION
 ^
/home/vagrant/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/esp_sdk_ver.h:1:4: error: stray '#' in program
 -e #undef ESP_SDK_VERSION
    ^
make: *** [build/ets_alt_task.o] Error 1
`

sanderboele, 2016-03-13:
I had this problem.

Edit this file: /home/vagrant/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/esp_sdk_ver.h

and make it look like this

```
#undef ESP_SDK_VERSION
#define ESP_SDK_VERSION 010502
```

dpgeorge, 2016-03-13:
This is an issue for esp-open-sdk and has been previously reported.  Please see https://github.com/pfalcon/esp-open-sdk/issues/136 and continue the discussion there, thanks!

closed 2016-03-13 by dpgeorge

bisguzar, 2016-03-13:
@sanderboele i cant found this file. where is it :/ i use "ls -a" for list folders. but i cant see this file. @dpgeorge