micropython/micropython #1895: task prio out of range: 2

sanderboele, 2016-03-13:
I just compiled the newest version for today, it compiles fine and I can write it to my nodeMCU 0.9 (ESP-12), but when I try to attacht a screen to the serial console I get this:
```
MicroPython v1.6-214-ge5c39a3-dirty on 2016-03-13; ESP module with ESP8266
Type "help()" for more information.
>>> #4 ets_task(40100470, 2, 3fff8a24, 16)
task prio out of range: 2
task prio out of range: 2
```
and then it locks up. What is this?

wendlers, 2016-03-13:
Got the same problem here with the ESP. Don't know why this happens, but when I change 

#define UART_TASK_ID 0

in  "esp8266/esp_mphal.h" to:

#define UART_TASK_ID 1

I at least get the REPL again.

sanderboele, 2016-03-13:
How the world did you figure that out?

it works though!

```
>>>
MicroPython v1.6-214-ge5c39a3-dirty on 2016-03-13; ESP module with ESP8266
Type "help()" for more information.
>>> #4 ets_task(40100470, 3, 3fff8a24, 16)
ets_alt_task: ets_run
q for task 3: queue: 3fff8a24, get ptr: 0, put ptr: 0, qlen: 16
q for task 20: queue: 3ffef780, get ptr: 0, put ptr: 0, qlen: 3
q for task 28: queue: 3fff93b0, get ptr: 0, put ptr: 0, qlen: 10
q for task 31: queue: 3ffeea30, get ptr: 0, put ptr: 0, qlen: 4
q for task 32: queue: 3ffef9f0, get ptr: 0, put ptr: 1, qlen: 34
====

>>>
```

wendlers, 2016-03-13:
To my understanding UART_TASK_ID defines the priority of the UART task. I just thought: if it is not happy with prio 0, let's try with 1. Still I don't know yet why it is not happy (any more) with prio 0 ... might have to do with the ESP SDK (but not sure).

dpgeorge, 2016-03-17:
Should be fixed by 556a1df6fd340a8dc6ededef2c5978391a7b965d.

closed 2016-03-17 by dpgeorge

wendlers, 2016-03-17:
Hi Damien,

it works on my setup too.

bisguzar, 2016-03-17:
@dpgeorge im using adafruit's build guide. where is this file? idk

dpgeorge, 2016-03-17:
> where is this file? idk

@bisguzar please ask a more detailed question.

bisguzar, 2016-03-18:
@dpgeorge owright, Im using this guide for build my own micropython.bin for nodemcu https://learn.adafruit.com/building-and-running-micropython-on-the-esp8266/build-firmware . I cant found esp_mphal.h or ets_alt_task.c . what can i do?

dpgeorge, 2016-03-19:
> I cant found esp_mphal.h or ets_alt_task.c . what can i do?

Those files are definitely in the repository, on the master branch.  Try making a fresh clone of the repo.

bisguzar, 2016-03-19:
@dpgeorge sorry for my basic questions but can i build it on virtualmachine? if i can i think buy a machine on digitalocean for build

dpgeorge, 2016-03-19:
> can i build it on virtualmachine?

Yes, but please ask such questions on https://forum.micropython.org, thanks!