micropython/micropython #1897: ESP8266 - OSError 12 after processing requests for a while with lwip sockets

wendlers, 2016-03-13:
I tried to implement a very simple server with lwip sockets. The test script below runs for a while as expected, but then terminates with OSError 12 (cannot allocate memory?). 

Versions used (one ESP826612E NodeMCU):

```
>>> import os
>>> os.uname()
(sysname='ESP8266', nodename='ESP8266', release='1.5.2(80914727)', version='v1.6-217-g99fc0d1-dirty on 2016-03-13', machine='ESP module with ESP8266')
```

Script used on the ESP:

```
import lwip

srv = lwip.socket()
srv.bind(("", 80))
srv.listen(0)

cnt = 0

print("server started")

reshdr = "HTTP/1.1 200 OK\r\nConnection: close\r\n"	

try: 

	while True:

		cli, addr = srv.accept()

		print(cnt, addr)
		cnt += 1

		inp = cli.read(128)
		cli.send("HTTP/1.1 200 OK\r\nConnection: close\r\n")	
		cli.close()

except OSError as e:

	print("Failed with OSError %s after %d requests" % (e, cnt))
```

Script used  on the client machine:

```
while true; do curl http://192.168.1.126/; done
```
The result of the script repeatedly is:

```
server started
0 ('192.168.1.113', 40546)
1 ('192.168.1.113', 40548)
2 ('192.168.1.113', 40550)
...
37 ('192.168.1.113', 40620)
38 ('192.168.1.113', 40622)
39 ('192.168.1.113', 40624)
Failed with OSError 12 after 40 requests
```

dpgeorge, 2016-03-17:
I can confirm this is a problem with the code from this repo (as at 2b302dad512fb3540db8b0b3a4b33337dd5d81a2) and also with the alpha version that was developed during the Kickstarter.  The latter firmware lasts up till request 137 then gives an "E:M" error which is indeed out of memory.  Once that code is fully merged we can revisit this problem.

wendlers, 2016-03-17:
OK. Looks too me like this is mainly caused by the "accept" thing. I tried a more "telnet" style variant, where the client only connects "once", and then does a lot of read-write. The read/write cycle seam to work "for ever" (almost, but at least much longer).

dpgeorge, 2016-03-17:
> OK. Looks too me like this is mainly caused by the "accept" thing.

Probably.  But it's actually failing in the cli.send line.

dpgeorge, 2016-03-17:
I investigated this a bit further.  With the "alpha" code, putting a "sleep 1s" between curls makes the problem go away (well, I can get up to 1000 iterations and then I got bored).  A "sleep 0.5s" does not make the problem go away.  And using "sleep 1s" with the code in this repo also does not fix the problem.

wendlers, 2016-03-18:
I could confirm this. With alpha v02 it runs longer (~130 times). Also the sleep in the curl helps. I added an output of the free memory from the gc module at the beginning and at the end like this:

```
        while True:

                cli, addr = srv.accept()

                print("free in: %d" % gc.mem_free())

                print(cnt, addr)
                cnt += 1

                inp = cli.read(128)
                cli.send("HTTP/1.1 200 OK\r\nConnection: close\r\n")    
                cli.close()

                print("free out: %d" % gc.mem_free())
```

Reported memory seams to get low when it fails:

```
free in: 16304
0 ('192.168.1.109', 48710)
free out: 16176
free in: 16048
1 ('192.168.1.109', 48712)
free out: 15920
free in: 15792
2 ('192.168.1.109', 48714)
free out: 15664
free in: 15536
3 ('192.168.1.109', 48716)
free out: 15408
free in: 15280
...
61 ('192.168.1.109', 48832)
free out: 560
free in: 432
62 ('192.168.1.109', 48834)
free out: 304
free in: 176
63 ('192.168.1.109', 48836)
free out: 16768
free in: 16640
...
125 ('192.168.1.109', 48960)
free out: 896
free in: 768
126 ('192.168.1.109', 48962)
free out: 640
free in: 512
127 ('192.168.1.109', 48964)
free out: 384
free in: 256
128 ('192.168.1.109', 48966)
E:M 1584
Failed with OSError 12 after 129 requests
```

But the first time (about #62/63) when memory gets low and GC is collecting, there is not problem. The second time, before the next collect, it fails. I don't know if the problem has to do with GC managed memory at all, since I also tired to do a manual "gc.collect()" right after "cli.close". this kept up the free memory but failed anyways after about 130 times ...

pfalcon, 2016-04-11:
So, there're many issues involved here, one particular issue is poor error reporting on lwIP side. It's listen function doesn't return error code, just NULL. Obvious cause would inability to alloc listening socket, and that's why modlwip translates that to ENOMEM. But it can other error, for example EADDRINUSE. And that's what actually happens here, even with SO_REUSEADDR, because lwIP implements some standard semantics for it, which doesn't make sense in our case. So, we need to think what makes sense in our case and patch lwip for it.

pfalcon, 2016-04-11:
The latter part is similar to https://github.com/micropython/micropython/issues/1971 , which has important information on how to interpret the effect being seen.

delivrance, 2018-10-01:
Here's a quick and dirty workaround that works for me.
The idea is simply wait until micropython itself frees enough memory. Waiting for 8 KB to be freed takes ~5 seconds:

```python
import socket
import esp
import time

server = socket.socket()
server.bind(("", 80))

server.listen(1)

while True:
    print(esp.freemem())

    if esp.freemem() < 1 * 1024:
        s = time.ticks_ms()

        while esp.freemem() < 9 * 1024:
            pass

        print("waited {} ms".format((time.ticks_ms() - s)))

    client, address = server.accept()
    data = client.recv(1024)
    client.sendall("{} {}".format(client, address).encode())
    client.close()
```

delivrance, 2018-10-02:
@peterhinch Yes, but doesn't seem to be helping, at all.

dpgeorge, 2019-07-03:
Should be fixed by 1d6cb6357a2cde40822712211fb51c4a4bbe41cf (see that commit for details).

With that patch applied the test script in comment https://github.com/micropython/micropython/issues/1897#issuecomment-398645097 handled 5000 hits of curl before I stopped it (it showed E:M but was able to recover).

closed 2019-07-03 by dpgeorge

arslan437, 2019-10-15:
Hi 
I am getting **### OSError: [Errno 12] ENOMEM** .  I have tried esp.freemem() its say esp has no fuction freemem. I'm using esp32. I know thread is of esp8266. If any have any idea my code is 
`
s = socket.socket()
    ai = socket.getaddrinfo(host, 443)
    addr = ai[0][-1]
    s.close()

    count = 0
    errorCount = 0
    while True:
        # if not oledDisplay:
        #     builtinLed.value(0)  # turn on led

        checkwifi()
        newSasToken()
        count = count + 1

        #temperature, pressure, humidity = mySensor.measure()
        temperature = 37
        pressure = 52
        humidity = 70
        light = 0
        freeMemory = gc.mem_free()

        print (gc.mem_free())

        data = b'{"Id":%u,"Mem":%u,"Celsius":%s,"hPa":%s,"Humidity":%s,  "Light":%d, "Errors":%d}' % (
            count, freeMemory, temperature, pressure, humidity,  light, errorCount)
        print (gc.mem_free())
        
        print ('trying')
        s = socket.socket()
        print(addr)
        s.connect(addr)
        
        s = ussl.wrap_socket(s)  # SSL wrap
        print ('sent')
        # Send POST request to Azure IoT Hub
        # s.write("POST /devices/" + cfg.deviceId +
        #         "/messages/events?api-version=2016-02-03 HTTP/1.0\r\n")
        
        s.write("POST " + hub.endpoint + " HTTP/1.1\r\n")
        # HTTP Headers
        s.write("Host: " + host + "\r\n")
        s.write("Authorization: " + SAS + "\r\n")
        s.write("Content-Type: application/json\r\n")
        s.write("Connection: close\r\n")
        s.write("Content-Length: " + str(len(data)) + "\r\n\r\n")
        # Data
        s.write(data)

        
        # Print 128 bytes of response
        print(s.read(128)[:12])

        s.close()
        time.sleep(sampleRate)
`