micropython/micropython #1900: ESP8266 MICROPYTHON BUILD error issue

skylin008, 2016-03-14:
I had upadte the latest version of esp-open-sdk and micropython for esp8266 ,but the other issues shows,follow the error message:
/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -lmain /opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -ljson /opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -lssl /opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -llwip_open /opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -lpp /opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -lnet80211 /opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -lwpa /opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -lphy /opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ld: cannot find -lnet80211 make:***[build/firmware.elf]] Error1.
 How to solve this problem ,TKS advance!

dpgeorge, 2016-03-17:
You need to provide more information to help debug this problem.  Use "make V=1" to build and see what the option to -L is.  Then see if that directory actually exists, and if it has the libraries in it.

pfalcon, 2016-03-23:
Please see http://forum.micropython.org/viewtopic.php?f=16&t=1689 for further instructions.

closed 2016-03-23 by pfalcon