micropython/micropython #1901: stmhal/makefile changed "default" USE_PYDFU ?=0 to 1

rolandvs, 2016-03-14:
Although minor or with good reason: In the ./stmhal/Makefile there seems to be a change being `USE_PYDFU ?= 1`, but I expect that the `dfu-util` is preferred in case you are working with the STM micro's?

dhylands, 2016-03-14:
pydfu.py is actually faster than dfu-util, and means that there is one less piece of software that the end user needs to install.

rolandvs, 2016-03-14:
In the case of using my Mac (OSX10.11.3) the usb.core and usb.util are not "standard" available in python 2.x/3.x. I stick to the dfu-util for now... try again later... :-)

dpgeorge, 2016-03-14:
As @dhylands points out, we now recommend to use pydfu.py, which is included in the tools/ directory of this repo.  You can stick to the old dfu-util by creating your own "makefile" (note lower case "m") in the stmhal directory, and putting into it the line: `USE_PYDFU=0`.

closed 2016-03-14 by dpgeorge

peterhinch, 2016-03-15:
@rolandvs To get pydfu.py running under Linux I had to execute 
```
$ sudo pip install pyusb --pre
```