micropython/micropython #1902: Junior issues

prusnak, 2016-03-15:
An easy way how developers can get introduced to the project is marking easy low-hanging fruits with some tag, e.g. "junior". That way developers can mark issues that are easy, but at the same time not very worth of senior developers' time. Newcomers can filter these issues and start digging in the code and at the same time help the project by sending fixes via PR.

dpgeorge, 2016-03-15:
We already have an issue for this, see #1331.  Problem is that you can't make issues "sticky" so they get lost very easily.

prusnak, 2016-03-15:
Sorry, missed #1331. Closing.

prusnak, 2016-03-15:
Sticky is indeed not possible. However, you can create a link to issue filter and put this link in README or other documents. 

https://github.com/micropython/micropython/issues?q=is:open+is:issue+label:bug

closed 2016-03-15 by prusnak

pfalcon, 2016-03-15:
We have a topic on the forum, and it is sticky: http://forum.micropython.org/viewtopic.php?f=3&t=779 . Maintenance of it is a separate issue though. There should be "generations" of developers, with the current one "caring" about next one. @dpgeorge being generation 0, was very welcoming of e.g. mine efforts (even though some of them were follies/dead ends). I'm in turn created that topic, which led to modlwip creation, which is important milestone for uPy. Well, we need more people to take the baton and pass it forward.