micropython/micropython #1906: inline asm: return value is zero if four arguments passed

peterhinch, 2016-03-16:
MicroPython v1.6-223-g77f85db on 2016-03-15; PYBv1.0 with STM32F405RG
```python
@micropython.asm_thumb
def foo(r0, r1, r2, r3):
    movw(r0, 25)
```
```
>>> foo(1,2,3,4)
0
>>> 
```
The fault only occurs if the function uses four arguments.

dhylands, 2016-03-16:
I seem to recall @dpgeorge saying that only 3 arguments were supported.

peterhinch, 2016-03-16:
Oops, I've just rediscovered this myself.

closed 2016-03-16 by peterhinch

dpgeorge, 2016-03-16:
As easy fix.  4 args to inline asm now implemented in 9a58316de235fed23566f9898de43b57fab85d7d.

peterhinch, 2016-03-16:
An error message would be good if too many args are supplied in an asm function def. Viper does this.

dpgeorge, 2016-03-16:
> An error message would be good if too many args are supplied in an asm function def

Already does: 8dfbd2d5894c98bd7b48e3e69bec0216d06ebb47

peterhinch, 2016-03-17:
The issue arose because the code failed silently, using firmware dated 15th March. Today's build is fine, accepting four args and producing an error if more appear in the function def. Great work!

dpgeorge, 2016-03-17:
The code to check that 4 or less args are used in an asm function has been there since the very beginning: a2f2f7db1f42f9229df9d4f353689f24685e5dc4.  But as you can see I was a bit confused as to what the maximum were: that commit checks and warns if 5 or more args are used, but the error message says at maximum 3 args are allowed.  I had it in my mind that 4 args would be ok, but then wasn't sure how the 4th arg was passed, on the stack or in a register.  If it was passed on the stack (by the calling C function) then it would require more effort to get working.  But it is indeed passed as a register so 4 args can (and is) supported.