micropython/micropython #1908: could not find module 'boot'

bisguzar, 2016-03-16:
Hello everyone, i have a problem. First im flashin micropython-firmware.bin to my nodemcu dev kit. (esp8266 based). Its finished, the firmware is flashed. Secon im testing connect my board on serial connection with python, and i cant see anything in console. Next im restarting my board and i saw some wrongs on putty console. it is:

Note: baudrate 115200

`rll▒▒|▒l▒|
▒
l▒
b|▒▒▒▒r▒b▒
b▒▒nn▒lnn▒▒▒
bp
lblrlp▒n▒
▒
l
b
n▒|▒
▒
▒
b▒▒nn▒l▒▒l`▒nnl▒l`nr▒▒▒n
▒b▒``▒n
▒▒b▒nl▒
▒▒nn▒
▒lp▒n▒
r▒▒▒▒▒
b
n▒|l▒
▒b▒▒nn▒l▒
l`▒nn
l`nr▒▒▒n
▒
▒lp▒n▒
r▒▒▒▒▒▒l
b
n▒|▒▒▒p⌎b▒▒nn▒
l`▒nnl▒l`nr▒▒▒n
bl`r▒▒n
bl`<
l▒▒|▒▒rrnb▒▒l
▒
▒b▒l
b쌜▒▒
b
▒lrldon't use rtc mem data
l
l▒▒|▒▒rrnb▒▒l▒▒p▒
▒b▒
b쌜▒▒
▒
b▒
▒lrl▒
l▒▒|▒▒rrnb▒▒ll▒p
b▒
b쌜▒▒▒
b▒lblrl▒▒rl▒▒▒b
l▒br|▒
b
ll
▒▒▒lb▒▒n▒nn▒
▒l▒|▒l
▒
l▒▒l▒▒▒▒▒▒l`▒n▒▒▒▒b
▒
▒b
▒▒▒b
l▒b▒lrlr▒n▒▒▒
▒▒n▒▒▒▒brb▒▒b쎜▒l▒▒
▒lbl▒▒ln▒prl▒
l▒▒|▒▒rrnb▒could not find module 'boot'

>>>
MicroPython v1.6-214-ge5c39a3 on 2016-03-13; ESP module with ESP8266
Type "help()" for more information.
>>> #4 ets_task(40100470, 2, 3fff8a04, 16)
task prio out of range: 2
task prio out of range: 2
`

dpgeorge, 2016-03-16:
Everything is working as it should.  The only problem is the "task prio out of range: 2", but see #1895 for that.

closed 2016-03-16 by dpgeorge

bisguzar, 2016-03-16:
@dpgeorge sorry but im not professional linux user. can you sen me your .bin file for nodemcu? if you have anyone

dpgeorge, 2016-03-16:
@bisguzar no, sorry, firmware is at the moment only available for certain backers of the Kickstarter (without the Kickstarter the ESP8266 code would not exist as it does now).  Issue #1895 should be resolved soon and then you can compile your own firmware just like you have already done.

bisguzar, 2016-03-17:
@dpgeorge then can i share my own compiled firmware file ? Is it legal ?

dpgeorge, 2016-03-17:
> can i share my own compiled firmware file ?

Please don't, that would go against the goals of the Kickstarter.  We have a great community and we try to uphold our values.

bisguzar, 2016-03-17:
I know. But i think record a video series for nodemcu with micropython. My viewers can not compile this project. Its not easy. What can i do ? Can i share with yout github link, name, project or more.

dpgeorge, 2016-03-17:
> But i think record a video series for nodemcu with micropython.

That sounds great!  But please note that the ESP8266 port is not yet ready for such advertisement.  It's only in alpha and lots of things don't work and/or are not implemented yet.  I would suggest you wait a month or so until there is a more stable version.  If you really want to do the series now, then you can link to github.com/micropython/micropython, micropython.org, forum.micropython.org.

bisguzar, 2016-03-17:
Thank you for this message and micropython. Have a nice day