micropython/micropython #1910: uzlib: support streaming

prusnak, 2016-03-17:
It would be nice if uzlib module supported streamed decompression. 

**Usecase:** Suppose I have a gzipped image file (PPM.GZ) stored on Flash/SD card. This file can be a few kilobytes big and I want to decompress its contents and dump into the LCD driver. Currently I cannot do this, because I have to decompress the whole file into main memory (using `zlib.decompress`), which is bad, because this data are often bigger than the whole RAM.

Or am I missing something?

pfalcon, 2016-03-17:
If this is a uzlib issue, why is it submitted against micropython? And yes, this is the main technical challenge towards implementing "upip on esp8266" stretch goal.

prusnak, 2016-03-17:
It is a missing feature of micropython. I don't know if uzlib supports this functionality and it is not exported via the module (maybe because it wants to copy API of zlib module from Python3) or it does not support it at all. That's why I am raising the point here.

peterhinch, 2016-03-18:
zlib does support it via zlib.decompressobj(wbits=15[, zdict]) see [Python docs](https://docs.python.org/3.5/library/zlib.html). The means of support is specifically designed for streams. This differs from the idea suggested by @prusnak - I have doubts about the feasibility of returning another file object. Random access to such an object would present challenges.

prusnak, 2016-03-18:
> Python docs[1]. The means of support is specifically designed for
> streams. This differs from the idea suggested by @prusnak[2] - I have

Yes, decompressobj is exactly what I meant.

 

Links:

  1. https://docs.python.org/3.5/library/zlib.html
  2. https://github.com/prusnak
  3. https://github.com/micropython/micropython/issues/1910#issuecomment-198215259

prusnak, 2016-03-26:
Moved to https://github.com/pfalcon/uzlib/issues/2

closed 2016-03-26 by prusnak