micropython/micropython #1912: stmhal: UART doesn't override RTS when buffer gets full

dhylands, 2016-03-20:
Currently (stmhal build) when using RTS HW flow control, the low level RTS signal seems to be behaving properly on a character by character basis. However, there is currently no interaction by the driver when the received buffer becomes full.

The uart driver should override RTS when the receive buffer becomes "full" (by whatever definition of full is appropriate - probably at least a few characters before the actual end of buffer). Once space is freed up in the buffer, then the uart driver should stop overriding RTS and let the HW control it again.

dpgeorge, 2016-03-20:
Maybe it's simpler to just not read the UART data register (uart->DR) if the buffer is "full", and in that case the MCU should just keep RTS asserted until uart->DR is read?

peterhinch, 2016-03-20:
I think RTS needs to be overridden from the outset. Currently when a single character is received RTS goes False and stays that way until the character is read. In other words it behaves as if the UART were unbuffered - i.e. how the hardware works. What is required is for RTS to remain True until the buffer is nearly full; this implies overriding the hardware. When the buffer is nearly full it should be asserted, either directly or by letting the hardware take control and not reading the DR.

One point which may or may not be relevant is that if an overrun occurs the hardware reasserts RTS - of course that's an error condition and data has been lost. But it seems odd to respond to an overrun by provoking further overruns.

Another possible bug concerns transmit timeouts. If CTS is False and you call UART.write() it immediately throws OSError 116 regardless of the value of timeout. I'd expect it to wait for whichever is first: the timeout expiring or CTS being asserted.

dhylands, 2016-03-20:
@peterhinch I disagree.

If RTS stays true, and a second character arrives before the first character is read then you've lost the first character. I'd agree with you if there was a HW FIFO, but there isn't. So the only time RTS should be overridden is when the receive buffer is nearly full, which prevents the hardware from saying its ok to send.

Damien's suggestion of leaving a character in the data register might work - I'd have to look at the IRQ implications (i.e would the IRQ keep firing). If the IRQ keeps firing then we'd need to figure out which is easier - managing the IRQs or mnaging the RTS.

peterhinch, 2016-03-21:
Point taken. But the way it's actually behaving seems to imply that, when flow control is enabled, the firmware isn't buffering received data. If it were, the arrival of a character would cause the hardware to set RTS False - but the f/w would read the char to buffer it, causing the h/w to set RTS True. This isn't happening. The system is behaving as if it were unbuffered.

dhylands, 2016-03-21:
sure - the hardware is unbuffered, and there's nothing we can do to change that, so it has to behave that way. Any scenario that leaves RTS true while the character is sitting in the register, is a scenario under which that character can be lost.

peterhinch, 2016-03-21:
I hope I'm not being obtuse but I'd expect the sequence to be as follows:

Char arrives. h/w sets RTS False and raises interrupt.
f/w services interrupt by buffering the char, removing it from the UART.
h/w responds to the DR read by setting RTS true.
This would repeat until the buffer becomes 'full', a timeout occurs or the user reads the data.

In practice RTS only gets re-asserted when the _user_ reads the character. Given that the read is successful I assume the f/w is buffering the character. It's behaving as if the UART is unaware the character has been read until the user does a read - although looking at the h/w manual I can't see how this can occur. Alas I find the code somewhat impenetrable.

peterhinch, 2016-03-23:
The behaviour of the UART read is a bit more complex than I first realised. It seems that the unbuffered mode I described only occurs if you have previously provoked an OSError 116.

If you issue (to a UART object previously used):
```python
ser.deinit()
ser.init(56700, flow=ser.RTS|ser.CTS)
```
and immediately send characters to it, the characters are correctly buffered. It works perfectly so long as the buffer is never allowed to fill. Alas if the buffer is filled, RTS is never set False even if further characters are sent. A read() returns the characters buffered, but the overrun is (inevitably) lost.

So I think there are two issues with the code. Firstly an OSError 116 is altering the UART mode, apparently to unbuffered: I haven't yet figured out how.

Secondly, as @dhylands observed in the root message, the uart_irq_handler() function is unconditionally reading the DR rather than, as @dpgeorge suggested, checking the buffer status and only reading it if the buffer will accept the character.

My previous comments were observations of its behaviour after an OSError 116.

peterhinch, 2016-03-24:
I'm working on a fix for this and #1913 - it works but needs more testing before I submit a PR.

@dhylands Regarding the receive buffer I think that 'full' should mean full. This is because the UART receiver hardware is based on the assumption that the transmitting device can stop within one character period. I can't see a need to make a more conservative assumption: the h/w de-asserts RTS as soon as a character arrives. If the DR is not read and another arrives an overrun will occur - so the assumption of rapid response seems implicit.

In my solution, if the receive buffer fills, the DR is not read (as per @dpgeorge's suggestion), and the interrupt is disabled. The h/w sets RTS False. The int is re-enabled when space becomes available, enabling reception to re-commence as early as possible - the ISR reads the DR and the h/w sets RTS so no override is required.

The UART write() method blocks until either the entire buffer has been sent or the timeout has elapsed (normally because the target has caused CTS to go False). In a timeout it returns the number of bytes transmitted, enabling the caller to re-send the missing characters if required. If it times out sending the first character, it returns ``None`` as at present. This maintains compatibility although 0 is arguably more logical.

dpgeorge, 2016-04-13:
Fixed, see above.

closed 2016-04-13 by dpgeorge