micropython/micropython #1913: stmhal: uart raises an exception OSError 116 CTS is asserted

dhylands, 2016-03-20:
It seems that when CTS HW flow control and the remote end has asserted CTS, then issuing a write causes an OSError to be raised.

For multibyte writes this is bad, since you have no idea how much of your data was actually written.

Shouldn't it at least wait for the timeout to expire? Currently, if you set the timeout to 4000 and do a single character write while CTS is asserted then the it immediately raises the OSError ETIMEDOUT.

Some forum discussion related to this can be found here:
http://forum.micropython.org/viewtopic.php?p=9562

dpgeorge, 2016-03-20:
When using HW flow control with the UART, uPy is simply exposing the ST HAL behaviour... so I'd guess it could do with some improvements like you suggest!

dpgeorge, 2016-04-13:
Fixed, see above.

closed 2016-04-13 by dpgeorge