micropython/micropython #1914: ESP8266

slush0, 2016-03-20:
Hi, I'm trying to connect to WIFI using latest sources, getting following output in REPL:

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30512, room 16 
tail 0
chksum 0x93
load 0x3ffe8000, len 924, room 8 
tail 4
chksum 0x94
load 0x3ffe83a0, len 26224, room 4 
tail 12
chksum 0x9d
csum 0x9d
<line of binary data> ldon't use rtc mem data
<three lines of binary data>could not find module 'boot'
>>> 
MicroPython v1.6-246-gb894551-dirty on 2016-03-20; ESP module with ESP8266
Type "help()" for more information.
>>> #5 ets_task(401004e8, 2, 3fff8a14, 16)
ets_alt_task: ets_run
q for task 2: queue: 3fff8a14, get ptr: 0, put ptr: 0, qlen: 16
q for task 20: queue: 3ffef770, get ptr: 0, put ptr: 0, qlen: 3
q for task 28: queue: 3fff9780, get ptr: 0, put ptr: 0, qlen: 10
q for task 29: queue: 3fff9848, get ptr: 0, put ptr: 0, qlen: 10
q for task 31: queue: 3ffeea20, get ptr: 0, put ptr: 1, qlen: 4
q for task 32: queue: 3ffef9e0, get ptr: 0, put ptr: 1, qlen: 34
====

Code is mostly copy&paste from your docs:

`def setup_wifi():
    print('Configuring WIFI...')
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    if wlan.isconnected():
        wlan.disconnect()

    print('connecting to network...')
    wlan.connect(conf.wifi_ssid, conf.wifi_password)
    while not wlan.isconnected():
        pass
    print('network config:', wlan.ifconfig())
`

skylin008, 2016-03-21:
hi,slush0.you can add the boot.py in esp8266 scripts directory.In boot.py file wirte the code:import main .And put run code in main.py.You should try it again.

slush0, 2016-03-21:
Yep, did it, but the problem did not go away.

slush0, 2016-03-21:
I created boot.py to get rid of that module error. However connecting to WIFI from REPL should work, right? (That was core of the bug report - those binary data printed to REPL seems strange).

dpgeorge, 2016-03-21:
> However connecting to WIFI from REPL should work, right?

Yes, and as far as I'm aware, it does work.  The output to the REPL is debugging code, but it should still be connecting.

> those binary data printed to REPL seems strange

They are from the underlying OS boot routine and can't be disabled.  They are harmless.

slush0, 2016-03-21:
Ok, thank you for clarification. So everything is fine. Thank you for your amazing work!

closed 2016-03-21 by slush0