micropython/micropython #1915: Correct way to boot from internal flash when SD card attached

ryannathans, 2016-03-22:
In my application I have an SD card that may be present on boot (and maybe not)

I want to run python code from internal flash regardless of if there's an SD card or not.

At the moment my solution is as follows.

In main.c commenting out lines 512, 518, 520, 522, 523 fix this but it's probably a bit of a hacky fix. 

I'll make a pull request for the feature once I know the correct way to attack it.

What should be done in order to achieve this functionality? A simple #if covering these lines of code and a #define in mpconfigboard.h?

peterhinch, 2016-03-22:
This can be done in Python - e.g. in main.py.
```python
import os
os.chdir('/flash')
import your_file
```

dhylands, 2016-03-22:
Another method (that works on stmhal):

Put a boot.py card on the sdcard that contains:
```
execfile('/flash/boot.py')
```
and create /sd/main.py that contains:
```
execfile('/flash/main.py')
```

ryannathans, 2016-03-22:
Excellent suggestions but I can't control what content the user puts on or deletes from the SD. Also for security purposes, it's not fantastic to give them the ability to execute code

dhylands, 2016-03-22:
I guess the advantage of the way it is now is that you can stick in an sd card and have it boot exactly what's on the sdcard regardless of whats on the internal flash.

So there are basically 2 different camps - one that wants /flash to override and one that wants /sd to override and we don't currently have a way of selecting between them.

ryannathans, 2016-03-22:
Yes. The purpose of this issue is to discuss the correct way of adding support for both "camps"

peterhinch, 2016-03-22:
The way it is now is great for development in that you can always boot a known good SD card and recover from crashing code in /flash. A makefile option to override this behaviour would be useful to some.

dhylands, 2016-03-22:
This is my take on doing it using a configurable option.

Put something like the following in stmhal/mpconfigport.h

```
#define MICROPY_DEFAULT_SD_BOOT_DIR "/sd"
```
and then replacing the:
```
 f_chdrive("/sd");
```
with
```
 f_chdrive(MICROPY_DEFAULT_SD_BOOT_DIR);
```
and putting
```
#undef MICROPY_DEFAULT_SD_BOOT_DIR
#define MICROPY_DEFAULT_SD_BOOT_DIR "/flash"
```
in your boards mpconfigboard.h file.

ryannathans, 2016-03-22:
In my tests just commenting that one like didn't do the trick. I had to comment the other two or three lines as mentioned in the original issue. I see what you mean though. Might add this in simply as you've described

ryannathans, 2016-04-20:
Implemented, see #1997

ryannathans, 2016-05-27:
See continued discussion on above PR.

dpgeorge, 2017-02-01:
I think a simple, runtime-configurable solution to this problem is to have a special file on the flash drive called "SKIPSD" which has the following behaviour:
- if internal /flash filesystem has a file called SKIPSD then the SD card is ignored completely, as if it weren't plugged in
- if there is no SKIPSD in /flash then the SD card is mounted if it is plugged in and used as the boot drive (same behaviour as we currently have)

Additional things, like being able to expose SD card via USB MSC even if it's not mount, can be added later for even more configurability.

ryannathans, 2017-02-01:
That's a good idea, in my application I would have SKIPSD file and Mount the sd card in boot.py for data use

dpgeorge, 2017-02-07:
The "SKIPSD" file was implemented in 9779c99317f229435c07f11fd18223956de77b41.

> I want to run python code from internal flash regardless of if there's an SD card or not.

This is the original request/issue and is now achievable by creating an empty file "/flash/SKIPSD".  If the file exists then the board boots from internal flash regardless of the presence or not of an SD card.

> The way it is now is great for development in that you can always boot a known good SD card and recover from crashing code in /flash.

It'll still work this way *unless* you have a file SKIPSD in /flash.  If you do have such a file then you are doing something advanced and should understand the consequences.  It's anyway still easy to get around crashing code in /flash by doing a safe boot.

closed 2017-02-07 by dpgeorge

rlnktt, 2018-12-24:
@dpgeorge How do I create one such file immediately after boot always?

Thanks in advance!!

rlnktt, 2019-01-31:
@dpgeorge I have put up my main.py into drivers folder and have placed a symbolic link inside the directory modules. I have made a custom hw based on stm32f405vgt6 and as of now, it doesn't has USB exposed to outside world. Only way in which code can be loaded into the controller is by means of icsp. I have sd card for logging various informations at different point in time. I just need to run a script from internal flash memory. So I have put up pyexec_frozen_module(main.py) in main.c and this seem to run good if I load firmware0.bin in 0x08000000 and firmware1.bin in 0x08020000. 

I also have a custom bootloader that looks for a bin file in sd card and parses it and loads the contents into the flash memory. Whenever I make some changes to addresses in the linker script and load the bootloader, firmware0 and firmware1 altogether, I could see only a REPL prompt and nothing seem to come up as expected. 

What am I missing...

rlnktt, 2019-01-31:
> That's a good idea, in my application I would have SKIPSD file and Mount the sd card in boot.py for data use

If you could demonstrate this, it would be great!!