micropython/micropython #1923: The user wiki

peterhinch, 2016-03-22:
Is there any interest in my making an attempt at tidying this up? I've long felt it looks rather amateurish.

I'd like to do something along the lines of the work @vitiral did on the dev wiki, adding a sidebar (I installed a minimal demo effort today) and generally tidying up the content. Along with changing it from a single long page to a few pages indexed by the sidebar.

To date my contributions to it have been additive as I'm unsure of the ethics of deleting other peoples' content - examples of where this is appropriate are where improved content is now in the official documentation. A link is surely preferable to possibly inaccurate, outdated and un-reviewed material. What does the team think?

There is also content which is long out of date and needs editing.

If there is a way to do this edit offline and submit it for review I'm all ears :)

dpgeorge, 2016-03-22:
Is the wiki used enough to justify it?  We have the forum and the official docs.  I do think a wiki is a good idea, but would like to know that it's getting good use.

Ideally I'd want to switch to a more substantial piece of wiki software, something Python based, so that it's easier to maintain.  Eg MoinMoin.

peterhinch, 2016-03-23:
Wikis strike me as the best place for posting user generated code - un-reviewed code has no place in the official docs and forum posts can get lost in the noise. They are my first port of call when looking for code on other platforms.

As to whether ours is well-used, and whether it might be more used if it contained accurate material and looked a little more 'loved', I've no idea. Given that I'm usually the one who reports outages perhaps I'm its sole user :) Is there any way to measure usage?

Maybe it would be a waste of effort.

closed 2016-03-23 by peterhinch