micropython/micropython #1928: Recent buffer overflow fix broke esp8266 testsuite

pfalcon, 2016-03-24:
Currently, esp8266 testsuite locks up on float/string_format_modulo.py . The culprit is cea6cf8a5e87127eccc8214cbffa2d4c8aae8527 , reverting it fixes the issue. So, there may be still off-by-one or some other issue there.

dpgeorge, 2016-03-29:
This should now be fixed by e1e7657277f0c3ea6a207fa7768110475a52c39c.  But whole esp8266 test suite still fails randomly due to lack of heap.

pfalcon, 2016-03-30:
Confirmed fixed.

closed 2016-03-30 by pfalcon

pfalcon, 2016-04-01:
> But whole esp8266 test suite still fails randomly due to lack of heap.

Turned out it was not heap issue, but event-driven REPL one. With push REPL (now in master), the testsuite passes in 16K (with 15K, current default in master, couple of test fail). I'm very happy about this discovery as well as about our gradual progress which allows us to pay attention to and re-investigate such issues. It allows to be confident about the development direction.