micropython/micropython #1935: stmhal make failed!

jason--liu, 2016-03-26:
Hi, guys! I was trying to compile micropython for pyboard. when i made, it was falling with this message:

`Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Create build-PYBV10/genhdr/pins.h
Traceback (most recent call last):
  File "boards/make-pins.py", line 6, in <module>
    import argparse
ImportError: No module named argparse
make: *** [build-PYBV10/genhdr/pins.h] Error 1
make: *** [build-PYBV10/genhdr/pins.h] Deleting file `build-PYBV10/pins_PYBV10.c'`
Then i tried "make V=1", it shows this message:
`Create build-PYBV10/genhdr/pins.h
python boards/make-pins.py --board boards/PYBV10/pins.csv --af boards/stm32f405_af.csv --prefix boards/stm32f4xx_prefix.c --hdr build-PYBV10/genhdr/pins.h --qstr build-PYBV10/pins_qstr.h --af-const build-PYBV10/genhdr/pins_af_const.h --af-py build-PYBV10/pins_af.py > build-PYBV10/pins_PYBV10.c
Traceback (most recent call last):
  File "boards/make-pins.py", line 6, in <module>
    import argparse
ImportError: No module named argparse
make: *** [build-PYBV10/genhdr/pins.h] Error 1
make: *** [build-PYBV10/genhdr/pins.h] Deleting file `build-PYBV10/pins_PYBV10.c'`
**It looks likes like something wrong with argparse.**
**Can anyone tell me how could i fix this problem? thanks**

peterhinch, 2016-03-26:
I'd say there's something wrong with the Python installation on your build PC: argparse is part of the Python standard library and should be installed by default. Do you have an old version of Python lacking this library? It was introduced in V2.7/V3.2.

jason--liu, 2016-03-26:
@peterhinch You are right. Although i installed python3.5 in my Ubuntu, I just checked my python version, it was still V2.6. So, I think I have more work to do. Anyway, Thanks a lot.

closed 2016-03-27 by pfalcon