micropython/micropython #1939: esp8266: the i2c freq argument is ignored

deshipu, 2016-03-27:
The ``freq`` argument to ``machine.I2C`` doesn't seem to have any effect. No matter what value I pass to it, the resulting signal on the CLK line looks exactly the same on the osciloscope. Looking at the source code at https://github.com/micropython/micropython/blob/master/esp8266/modpybi2c.c#L84-L130 I can see that the "freq" variable is never used anywhere.

I tried to lower the frequency, because for some reason I'm getting occasional OSError when trying to write to the I2C bus, and, what's worse, those errors sometimes leave the device I'm communicating with in a stuck state.

dpgeorge, 2016-03-29:
Yes, changing the freq is not yet implemented.

deshipu, 2016-03-30:
By the way, on pyboard that argument is called "baudrate", perhaps it would be good to keep it consistent? http://docs.micropython.org/en/latest/pyboard/library/pyb.I2C.html

dpgeorge, 2016-04-13:
Setting the frequency is now implemented in d083712224602372c5fd22d0b9e2b32efe6fa26b (and then a525493e40eb4d89e805737bfa36b39d35221efb).

Yes, we need to decide on freq vs baudrate for the arg name.  But that's a separate discussion (along with all other discussions about new HW API).

closed 2016-04-13 by dpgeorge