micropython/micropython #1943: no esp.wifi_mode attribute is present

ceremcem, 2016-03-29:
As the error output says, there is no esp.wifi_mode attribute is present:
```
>>> esp.wifi_mode
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: no such attribute
```

Why is it giving this error?

dpgeorge, 2016-03-29:
The wifi_mode function was recently removed.  Please use network.WLAN.active instead.  See 679fe0abae92c6dc503b9ebc4a3b594a8b1f6aeb

closed 2016-03-29 by dpgeorge

vfl68, 2016-03-29:
First, wifi_mode seems to have moved to network, and second it's a function
so you need to call
`>>> network.wifi_mode()`

ceremcem, 2016-03-29:
I reached [the documentation](https://docs.micropython.org/en/v1.6/pyboard/library/esp.html) by [this](https://www.google.com.tr/search?q=micropython+esp&ie=utf-8&oe=utf-8&gws_rd=cr&ei=vP76VuajIsuisgGGsJP4Bg) google search. Would you please consider updating the documentation accordingly?

dpgeorge, 2016-03-29:
Please use the documentation for the latest version: https://docs.micropython.org/en/latest/pyboard/library/esp.html

We can't change the links that google gives...

ceremcem, 2016-03-29:
Okay. So would you give any documentation page link related to this problem? Because I can't find anything related to "wifi" in that page.  (Maybe you should consider placing a link to the documentation in [README.md](https://github.com/micropython/micropython/blob/master/esp8266/README.md) file.)

Additionally, I updated micropython to the latest commit (259967238423f2b11479dadbab6eae77cd038096) and rebuilt/flashed the esp8266 binaries. There seems no `network.WLAN.active` attribute: 

```
>>> import network
>>> dir(network)
['__name__', 'WLAN', 'wifi_mode', 'phy_mode', 'STA_IF', 'AP_IF', 'STAT_IDLE', 'STAT_CONNECTING', 'STAT_WRONG_PASSWORD', 'STAT_NO_AP_FOUND', 'STAT_CONNECT_FAIL', 'STAT_GOT_IP']
>>> network.WLAN.active
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: no such attribute
```

pfalcon, 2016-03-30:
@ceremcem , the algorithm is:

1. Go to http://docs.micropython.org/
2. Select the right port in "Ports and Version" popup menu on left bottom.
3. Follow documentation thoroughly.

E.g. http://docs.micropython.org/en/latest/esp8266/esp8266/quickref.html makes clear how it should be used:

~~~
wlan = network.WLAN(network.STA_IF)
~~~

Also keep in mind that ESP8266 port at this point isn't suitable for end users, it is shared with developers who are interested in MicroPython [and are ready to delve in the source code and git logs]. Find more information on the forum: http://forum.micropython.org/viewtopic.php?f=16&t=1689

ceremcem, 2016-03-30:
Thank you for your further explanations warning about end users. 

By the way, here is the interpreter outputs I could follow: 

```
print("hello there")
hello there
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> wlan.scan()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: argument num/types mismatch
>>> 
```

pfalcon, 2016-03-30:
wlan.scan() issue fixed.

ceremcem, 2016-03-30:
Yes, it is fixed: 

```
print("naber")
naber
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> wlan.scan()
f r0, scandone
[(b'aktos-elektronik', b'\x00\x1c\xa8\xf7\xe8M', 6, -49, 3, 0), (b'UEM', b'\xf4\xe3\xfb\xbak\xdc', 9, -92, 4, 0), (b'TTNET_HUAWEI_827F', b'\xec#=d\x82\x88', 10, -91, 4, 0)]
>>> wlan.isconnected()
False
>>> wlan.connect("aea", "084DA789BF")
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4000deed, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40000008, depc=0x00000000

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)
```