micropython/micropython #1944: V1.6 USB REPL broken on STM32F429DISC

mulleand, 2016-03-30:
I can still see the Pyboard USB Comm Port in the device manager under windows and it's also possible to access the file system but I cannot get the REPL working. Everything worked fine with 1.5.2.

I tested it with two different boards under Windoes with Putty as well as Linux with screen.

vfl68, 2016-03-30:
I've installed the v1.6-308-g6249da8 on the pyboard. I can use the repl from linux and from windows
On windows I use KiTTY, but I had to install a driver with the pybcdc.inf file.
Perhaps there are some differences between the serial drivers, did you try to reinstall it ?

mulleand, 2016-03-31:
It works on my pyboard v1.1 just not on the STM32F429DISC-board. I also reinstalled the driver but that didn't help

hoihu, 2016-04-03:
The STM429DISC has the USB configured using HS mode using FS( "USE_USB_HS"). The SOF irqs that were introduced in 1.6 are at the moment only configured in pure FS mode hence the callback that inputs the REPL with the chars received from USB is not called.

So 2 options: 
set "Sof_enable=1" on line 432 in usbd_conf.c

or use USE_USB_FS in your hal configure file - i'm not sure why it was configured to use HS mode though, because all other ports seem to use FS mode.

dhylands, 2016-04-03:
The 429DISC board uses HS mode because that's what the USB connector on the board connects to. The pins that the FS USB signals are on are being used by the LCD display circuitry.

hoihu, 2016-04-03:
ok - so best to try sof_enable=1 then.

mulleand, 2016-04-04:
Thanks hoihu and dhylands for your help. I followed your advice and it works as expected. 
Would it be possible to merge this change into the next micropython release?

hoihu, 2016-04-04:
You can place a pull request then it should find its way to the next release.

If that is too much for you then I can do that, let me know..

mulleand, 2016-04-05:
Sorry guys. I made a mess here. I will try it again

commit 8a8e7750 (2016-04-07): stmhal: Consistently enable USB SOF Irqs for all USB modes (FS and HS).

SOF irqs are now standard for rx/tx USB transfers, so enable them for both
FS and HS modes.  Fixes #1944.
closed 2016-04-07 by dpgeorge