micropython/micropython #1946: wlan.connect function causes esp8266-03 to reset

ceremcem, 2016-03-30:
As a follow up of #1943, `wlan.connect` function causes reset: 

```
print("naber")
naber
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> wlan.scan()
f r0, scandone
[(b'aktos-elektronik', b'\x00\x1c\xa8\xf7\xe8M', 6, -49, 3, 0), (b'UEM', b'\xf4\xe3\xfb\xbak\xdc', 9, -92, 4, 0), (b'TTNET_HUAWEI_827F', b'\xec#=d\x82\x88', 10, -91, 4, 0)]
>>> wlan.isconnected()
False
>>> wlan.connect("aea", "084DA789BF")
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4000deed, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40000008, depc=0x00000000

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)
```

pfalcon, 2016-03-30:
Can't reproduce.

deshipu, 2016-03-30:
I still get this with the newest build (with the https://github.com/micropython/micropython/commit/d88250c06eb9f2f56b790c1411718134a148d9f4 commit included):

```
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> wlan.scan()
f r0, scandone
[(b'swamp', b'\x8c\x04\xffz\xabN', 1, -60, 4, 0), (b'RAVEENDRAN-HP_Network', b'\\\xd9\x98!mJ', 1, -84, 3, 0), (b'FRITZ!Box Fon WLAN 7390', b'$e\x11\xf7\xba\xed', 1, -91, 4, 0), (b'UPC247179715', b'T\x88\x0e\x1b\xf2\xd8', 6, -73, 3, 0), (b'alfl', b'\x00\x01\xe3\xd0\x96\x1e', 6, -75, 1, 0), (b'HZN243233587', b'T\xfa>"\xdb\x03', 7, -60, 3, 0), (b'UPC Wi-Free', b'V\xfa>"\xdb\x03', 7, -59, 5, 0), (b'qyk-70415', b'\xa8\xd3\xf7A\x86\xf8', 11, -77, 4, 0), (b'UPC243000456', b'T\x88\x0e-\xb6H', 11, -63, 3, 0), (b'UPC Wi-Free', b'T\x88\x0e-\xb6I', 11, -60, 5, 0), (b'hhf-59716', b'\\\xdc\x96\xbbj\x9b', 11, -87, 4, 0), (b'HZN246302697', b'T\xfa>,R_', 13, -72, 3, 0), (b'UPC Wi-Free', b'V\xfa>,R_', 13, -75, 5, 0)]
>>> wlan.connect('swamp', 'beepbeep')
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4000deed, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40000060, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 31664, room 16 
tail 0
chksum 0xe3
load 0x3ffe8000, len 996, room 8 
tail 12
chksum 0x5c
ho 0 tail 12 room 4
load 0x3ffe83f0, len 29800, room 12 
tail 12
chksum 0x52
csum 0x52
|l��|��rrnb��ll�b�lb쌜���b�lrldon't use rtc mem data
ll��|��rrnb����l�b�b쌜���b��lrl�l��|��rrnb��l����b�b쌜��l�p�b�lblrl��rl���bl�br|�bll���lb��n�nn��l�|�l�l��l������l`�n����b��b���bl��could not find module 'boot'

>>> 
MicroPython v1.6-327-g61fa7c8-dirty on 2016-03-30; ESP module with ESP8266
Type "help()" for more information.
>>> ets_alt_task: ets_run
q for task 2: queue: 3fff9668, get ptr: 0, put ptr: 0, qlen: 16
q for task 20: queue: 3fff05c0, get ptr: 0, put ptr: 0, qlen: 3
q for task 28: queue: 3fffa2d8, get ptr: 0, put ptr: 0, qlen: 10
q for task 31: queue: 3ffef870, get ptr: 0, put ptr: 0, qlen: 4
q for task 32: queue: 3fff0830, get ptr: 0, put ptr: 0, qlen: 34
====

>>> 
```

pfalcon, 2016-03-30:
That's crazy guys. Are you sure you use the latest esp-open-sdk (make clean; make in it)? Are you sure you tried esptool.py erase_flash? Then please try to reduce heap size even further (a-la d88250c linked above). (And yep, we have solution for heap size, we'd like to test more and more and more before dumping it to master, because otherwise it can be support hazard).

deshipu, 2016-03-30:
I'm trying with a clean sdk now, but even running "make clean" takes half an hour, so that will take some time.

deshipu, 2016-03-31:
So I compiled everything from scratch, both sdk and the micropython software. I did erase_flash. I got this:

```
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
mode : sta(18:fe:34:d3:7f:a4) + softAP(1a:fe:34:d3:7f:a4)
#5 ets_task(40214e78, 28, 3fffad60, 10)
add if0
>>> wlan.connect('swamp', 'beepbeep')
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4000deed, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40000040, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 31664, room 16 
tail 0
chksum 0xe3
load 0x3ffe8000, len 996, room 8 
tail 12
chksum 0x5c
ho 0 tail 12 room 4
load 0x3ffe83f0, len 29800, room 12 
tail 12
chksum 0x52
csum 0x52
|l��|��rrnb��ll�b�lb쌜���b�lrldon't use rtc mem data
ll��|��rrnb����l�b�b쌜���b��lrl�l��|��rrnb��l����b�b쌜��l�p�b�lblrl��rl���bl�br|�bll���lb��n�nn��l�|�l�l��l������l`�n����b��b���bl�b�lrlr�n����bp������r�쎜�lb�b��n����bpb������r�쎜�lbl�l��|��rrnb��l��b�b쌜���lbllbl��ln�l`��ln�prl�l��|��rrnb��l��b�lb쌜�����could not find module 'boot'

>>> 
MicroPython v1.6-327-g61fa7c8-dirty on 2016-03-31; ESP module with ESP8266
Type "help()" for more information.
>>> ets_alt_task: ets_run
q for task 2: queue: 3fff9668, get ptr: 0, put ptr: 0, qlen: 16
q for task 20: queue: 3fff05c0, get ptr: 0, put ptr: 0, qlen: 3
q for task 28: queue: 3fffa698, get ptr: 0, put ptr: 0, qlen: 10
q for task 29: queue: 3fffa760, get ptr: 0, put ptr: 0, qlen: 10
q for task 31: queue: 3ffef870, get ptr: 0, put ptr: 1, qlen: 4
q for task 32: queue: 3fff0830, get ptr: 0, put ptr: 0, qlen: 34
====
```

This is an original Adafruit HUZZAH Feather board.

deshipu, 2016-03-31:
If I'm reading this right

```
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
mode : sta(18:fe:34:d3:7f:a4) + softAP(1a:fe:34:d3:7f:a4)
```
it gets into STA+AP mode even though I told it to go into STA mode?

deshipu, 2016-03-31:
I tried with:
```
STATIC char heap[14336];
```
and still get the same traceback.

deshipu, 2016-03-31:
I just noticed a weird discrepancy:
```
>>> import network
>>> import esp
>>> esp.STA_MODE
1
>>> network.STA_IF
0
```
Perhaps the constants didn't get changed everywhere?

pfalcon, 2016-03-31:
> it gets into STA+AP mode even though I told it to go into STA mode?

After "factory reset" (i.e., complete flash erase) SDK starts with AP interface active, then you enable STA interface.

pfalcon, 2016-03-31:
> STATIC char heap[14336];

Weird, very weird. The exception is the same as was originally reported, and decreasing heap size (to increase OS heap size) fixed it for me. One could explain these variations by different numbers of scan results (mine are pretty modest, apparently due to antenna performance, as I have pretty much a WiFi hell around me). It's still weird +1K didn't help. Please keep trying to decrease it. E.g., try 12000. And - you don't experience these problems with early release firmware, do you?

deshipu, 2016-03-31:
With 1200 it seems to work:
```
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.connect("swamp", "beepbeep")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: Cannot set STA config
>>> wlan.active(True)
mode : sta(18:fe:34:d3:7f:a4) + softAP(1a:fe:34:d3:7f:a4)
#5 ets_task(40214e78, 28, 3fff9fc0, 10)
add if0
>>> wlan.connect("swamp", "beepbeep")
>>> scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with swamp, channel 1
dhcp client start...

>>> ip:192.168.0.12,mask:255.255.255.0,gw:192.168.0.1
```

And yes, it did work with the alpha firmware, IIRC, I will try to make sure.

deshipu, 2016-03-31:
On alpha v2:
```
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
mode : sta(18:fe:34:d3:7f:a4) + softAP(1a:fe:34:d3:7f:a4)
#6 ets_task(4020e1d4, 28, 3fff5de0, 10)
add if0
>>> wlan.connect("swamp", "beepbeep")
>>> scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with swamp, channel 1
dhcp client start...
ip:192.168.0.12,mask:255.255.255.0,gw:192.168.0.1

>>>
```

deshipu, 2016-03-31:
Is there any way to make it run in STA mode only, without the AP?

pfalcon, 2016-03-31:
Thanks for confirming that it's OS heap issue.

> Is there any way to make it run in STA mode only, without the AP?

Of course, just deactivate AP interface (AP_IF).

deshipu, 2016-03-31:
I did a little bit of binary search, and determined, that it works with 13152, but fails with 13184. Didn't have the patience to narrow it down further.

slzatz, 2016-03-31:
On a fresh build with latest micropython and esp-open-sdk on Feather Huzzah, and with heap at 15360, I get the following:

    >>> import network
    >>> wlan = network.WLAN(network.STA_IF)
    >>> wlan.active(True)
    >>> wlan.connect(ssid, pw)
    Fatal exception 3(LoadStoreErrorCause):
    epc1=0x4000deed, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40000034,   depc=0x00000000

    ets Jan  8 2013,rst cause:2, boot mode:(1,7)

ceremcem, 2016-04-01:
I started to change the heap size to several values and flashed the firmware. 

My test automation [code is as follows](https://github.com/aktos-io/aktos-dcs/blob/master/examples/test_micropython.py):
```
from aktos_dcs import *

import sys

class TestMicropython(SerialPortReader):
    def prepare(self):
        print "started reading serial port..."
        print "-------------------------------"
        print self.ser

    def send_cmd(self, cmd, s=0.5):
        self.ser_write(cmd + "\r\n")
        sleep(s)

    def on_connect(self):
        print "Connected..."
        sleep(5)
        self.send_cmd("""import network""")
        self.send_cmd("""wlan = network.WLAN(network.STA_IF)""")
        self.send_cmd("""wlan.active(True)""")
        self.send_cmd("""wlan.scan()""", 5)
        self.send_cmd("""wlan.isconnected()""")
        self.send_cmd("""wlan.connect("aea", "084DA789BF")""", 10)
        self.send_cmd("""wlan.isconnected()""")
        self.send_cmd("""wlan.isconnected()""", 2)
        self.send_cmd("""wlan.isconnected()""", 2)
        self.send_cmd("""wlan.isconnected()""", 2)

    def serial_read(self, data):
        #print 'Serial Port Received: ', data
        sys.stdout.write(data)

TestMicropython(port="/dev/ttyUSB0", baud=115200)
wait_all()
```

> "Continuous output" means "the combined outputs of serial port while plugging and unplugging the device"

# Crash

Above approximately heap size of 12000, esp keeps crashing upon calling `wlan.connect()`. 

# Weird output

At heap size of 15000, device's continuous output is as follows: 
```
started reading serial port...
-------------------------------
Serial<id=0x7fcea42244d0, open=False>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False)
Trying to connect...
Connected...
import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> wlan.scan()
scandone
[(b'aktos-elektronik', b'\x00\x1c\xa8\xf7\xe8M', 6, -54, 3, 0)]
>>> wlan.isconnected()
True
>>> wlan.connect("aea", "084DA789BF")
>>> scandone
wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> Trying to connect...
Connected...
Trying to connect...
Connected...
Trying to connect...
Connected...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with aea, channel 1
dhcp client start...
wlan.isconnected()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name not defined
>>> wlan.connect("aea", "084DA789BF")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name not defined
>>> ip:192.168.2.128,mask:255.255.255.0,gw:192.168.2.101
import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> wlan.scan()
scandone
[(b'aktos-elektronik', b'\x00\x1c\xa8\xf7\xe8M', 6, -55, 3, 0), (b'TTNET_HUAWEI_827F', b'\xec#=d\x82\x88', 10, -94, 4, 0)]
>>> wlan.isconnected()
True
>>> wlan.isconnected()
True
>>> wlan.isconnected()
True
>>> wlan.connect("aea", "084DA789BF")
>>> scandone
wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> wlan.isconnected()
False

```

It's weird that `wlan.isconnected()` returns `True` **before** `wlan.connect()` and `False` **after** `wlan.connect()`. 

It's also weird that it connects the wireless network right after toggling the power while there is no file written permanently on the device.

slzatz, 2016-04-01:
For me, the problem persists on Feather Huzzah with heap = 13152 and with heap = 12000:

    >>> wlan.connect(ssid, pw)
    Fatal exception 3(LoadStoreErrorCause):
    epc1=0x4000deed, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40000008, depc=0x00000000

    ets Jan  8 2013,rst cause:2, boot mode:(1,7)

pfalcon, 2016-04-01:
@slzatz : Obvious next step is trying 10000, 8000, etc.

pfalcon, 2016-04-01:
> It's also weird that it connects the wireless network right after toggling the power while there is no file written permanently on the device.

@ceremcem : Test automation is good, but I figure you don't have much first-hand experience with esp8266 yourself. Please start with downloading docs (there're ~20 PDFs) from bbs.espressif.com and reviewing them. You will know (among a lot of other important things) that underlying SDK functions do save connection parameters persistently and a device reconnects on boot using them. Everyone who uses esp8266 know that and take for granted. (The latest SDKs have non-persistent versions, but I don't think we'll dare to switch to them as outcry will follow; I consider it as a possibility though ;-) ).

ceremcem, 2016-04-01:
@pfalcon Yes, I don't have much experience this device. But I do have experience with similar devices used in industrial automation. So I don't expect a device to do something if I don't say it to do so. 

I updated the micropython to the latest commit, discarded my heap setting and tested again. Here is the output:
```
started reading serial port...
-------------------------------
Serial<id=0x7fa4f59f14d0, open=False>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False)
Trying to connect...
Connected...
Trying to connect...
Connected...
import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> wlan.scan()
scandone
[(b'aktos-elektronik', b'\x00\x1c\xa8\xf7\xe8M', 6, -63, 3, 0)]
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> wlan.scan()
scandone
[(b'aktos-elektronik', b'\x00\x1c\xa8\xf7\xe8M', 6, -59, 3, 0)]
>>> wlan.isconnected()
False
>>> wlan.connect("aea", "084DA789BF")
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4000deed, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40000058, depc=0x00000000

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 31980, room 16 
tail 12
chksum 0x68
ho 0 tail 12 room 4
load 0x3ffe8000, len 996, room 12 
tail 8
chksum 0x23
load 0x3ffe83f0, len 29828, room 0 
tail 4
chksum 0x29
csum 0x29

l��|��rrnbl
            l�b�l
                 b쌜��l

                         b
                          rll
                             l��|��rrnb�
                                         �lb��
                                              b쌜��
                                                   lb�
                                                      rl�
                                                         l��|��rrnbl���
                                                                         �

                                                                          b쌜��l�
                                                                                 b�lblrlrl����b
                                                                                               l�br|�
                                                                                                     b
                                                                                                      ll
                                                                                                        �lb��n��nn�
         �l�|�l
                 �
                  l������l`�n�����b

                                     ����b
                                            l�blrlr�n���
                                                        ��bp�������
                                                                     ����쌜l�b��n����bpb�����
                                                                                                ����쌜lrl�
                                                                                                           l��|��rrnbl
         �

          b�

            b쌜����
                   �
                    bl
                      lbl��ln�l`��ln�prl
                                             ��|��rrnbl

                                                        
                                                         b�l
                                                            b쌜����l�could not find module 'boot'

>>> 
MicroPython v1.6-332-g777232c on 2016-04-01; ESP module with ESP8266
Type "help()" for more information.
>>> #6 ets_task(40100440, 3, 3fff9588, 4)
ets_alt_task: ets_run
q for task 2: queue: 3fff96a8, get ptr: 0, put ptr: 0, qlen: 16
q for task 3: queue: 3fff9588, get ptr: 0, put ptr: 0, qlen: 4
q for task 20: queue: 3fff05e0, get ptr: 0, put ptr: 0, qlen: 3
q for task 28: queue: 3fffa6d8, get ptr: 0, put ptr: 0, qlen: 10
q for task 29: queue: 3fffa7a0, get ptr: 0, put ptr: 0, qlen: 10
q for task 31: queue: 3ffef890, get ptr: 0, put ptr: 1, qlen: 4
q for task 32: queue: 3fff0850, get ptr: 0, put ptr: 0, qlen: 34
====
wlan.isconnected()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name not defined
>>> wlan.connect("aea", "084DA789BF")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name not defined
>>> wlan.isconnected()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name not defined
>>> wlan.isconnected()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name not defined
>>> wlan.isconnected()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name not defined
>>> wlan.isconnected()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name not defined
>>> wlan.isconnected()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name not defined
>>> wlan.isconnected()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name not defined
>>> wlan.isconnected()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name not defined
>>> wlan.isconnected()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name not defined

```

gschmottlach, 2016-04-02:
I was finally able to connect to my WLAN with the heap set to 11KB. Since this appears to be potentially RAM related, are there any plans to address this issue now that the basic unit tests are passing for the ESP8266 release? Seems like quite a few of us are suffering from this problem.

slzatz, 2016-04-02:
I was also able to connect to a wifi AP with the heap reduced to 11k.

ceremcem, 2016-04-02:
I don't know if I could connect the wifi or not, please take a look at the output (`heap[11000]`): 

```
ceremcem@cca-erik:examples$ python test_micropython.py 
started reading serial port...
-------------------------------
Serial<id=0x7fb89ed2d510, open=False>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False)
Trying to connect...
Connected...
import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> wlan.scan()
sl
scandone
usl
[(b'aktos-elektronik', b'\x00\x1c\xa8\xf7\xe8M', 6, -64, 3, 0)]
>>> wlan.isconnected()
True
>>> wlan.connect("aea", "084DA789BF")
sl
>>> scandone
usl
wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> Trying to connect...
Connected...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with aea, channel 1
dhcp client start...
ip:192.168.2.128,mask:255.255.255.0,gw:192.168.2.101
import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> wlan.scan()
scandone
[(b'aktos-elektronik', b'\x00\x1c\xa8\xf7\xe8M', 6, -76, 3, 0)]
>>> pm open,type:2 0
wlan.isconnected()
True
>>> wlan.connect("aea", "084DA789BF")
>>> sl
scandone
usl
wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> wlan.isconnected()
False
```

pfalcon, 2016-04-02:
@ceremcem

>  But I do have very much experience with similar devices used in industrial automation. So I don't expect a device to do something if I don't explicitly say it to do so.

Then you understand why I consider switching away form current "persistent" approach. As I mentioned, it unlikely to happen though. But if you'd like to try to influence that, feel free to start a topic on the forum, asking what other people think, and arguing your cause.

pfalcon, 2016-04-02:
All: please try the latest master (without decreasing the heap size).

ceremcem, 2016-04-02:
Note that `make clean && make` and `make` gives different `sha1sum` after `git stash && git pull`. I'll try `make clean && make` output.

ceremcem, 2016-04-02:
```
ceremcem@cca-erik:examples$ python test_micropython.py 
started reading serial port...
-------------------------------
Serial<id=0x7f4dcb8bcf50, open=False>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False)

[[ Device Physically Connected... ]]

import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
mode : sta(18:fe:34:d6:73:3f) + softAP(1a:fe:34:d6:73:3f)
#5 ets_task(402150d8, 28, 3fff3a00, 10)
add if0
>>> wlan.scan()
scandone
[(b'aktos-elektronik', b'\x00\x1c\xa8\xf7\xe8M', 6, -64, 3, 0)]
>>> wlan.isconnected()
False
>>> wlan.connect("aea", "084DA789BF")
>>> scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with aea, channel 1
dhcp client start...
ip:192.168.2.128,mask:255.255.255.0,gw:192.168.2.101
wlan.isconnected()
True
>>> wlan.isconnected()
True
>>> wlan.isconnected()
True
>>> wlan.isconnected()
True
>>> 
[[ Device Physically Disconnected... ]]

[[ Waiting for Device to be physically connected... ]]

[[ Device Physically Connected... ]]

[[ Device Physically Disconnected... ]]

[[ Waiting for Device to be physically connected... ]]

[[ Device Physically Connected... ]]

wlan.isconnected()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name not defined
>>> scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with aea, channel 1
dhcp client start...
wlan.isconnected()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name not defined
>>> ip:192.168.2.128,mask:255.255.255.0,gw:192.168.2.101
import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> wlan.scan()
scandone
[(b'aktos-elektronik', b'\x00\x1c\xa8\xf7\xe8M', 6, -64, 3, 0)]
>>> wlan.isconnected()
True
>>> wlan.connect("aea", "084DA789BF")
>>> scandone
wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> 
[[ Device Physically Disconnected... ]]

[[ Waiting for Device to be physically connected... ]]

[[ Device Physically Connected... ]]

scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with aea, channel 1
dhcp client start...
ip:192.168.2.128,mask:255.255.255.0,gw:192.168.2.101
import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> wlan.scan()
scandone
[(b'aktos-elektronik', b'\x00\x1c\xa8\xf7\xe8M', 6, -65, 3, 0)]
>>> wlan.isconnected()
True
>>> wlan.connect("aea", "084DA789BF")
>>> scandone
wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> wlan.isconnected()
False
>>> ^CException KeyError: KeyError(139972164226576,) in <module 'threading' from '/usr/lib/python2.7/threading.pyc'> ignored
ceremcem@cca-erik:examples$ ping 192.168.2.128
PING 192.168.2.128 (192.168.2.128) 56(84) bytes of data.
64 bytes from 192.168.2.128: icmp_seq=1 ttl=255 time=319 ms
64 bytes from 192.168.2.128: icmp_seq=2 ttl=255 time=8.15 ms
64 bytes from 192.168.2.128: icmp_seq=3 ttl=255 time=9.72 ms
64 bytes from 192.168.2.128: icmp_seq=4 ttl=255 time=6.62 ms
^C
--- 192.168.2.128 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3004ms
rtt min/avg/max/mdev = 6.622/85.985/319.435/134.786 ms
```

Note that `wlan.isconnected()` is returning `False` if `wlan.connect()` is called even though `wlan.isconnected()` returns `True`. 

Except for that, everything seems to work as expected.

gschmottlach, 2016-04-02:
I resync'ed and tried with a 16KB heap. This time it connected to my AP **without** a reset. Looks like Paul might've fixed this problem.

Now if I can only get the file-system to work . . . ;-)

Thanks for resolving this . . .

deshipu, 2016-04-03:
I can confirm that the newest firmware, MicroPython v1.6-353-g9698a60, connects to my WiFi without resets and without having to change the heap size.

pfalcon, 2016-04-03:
Thanks for reports. Please don't forget to run testsuite on every occasion ;-) : http://forum.micropython.org/viewtopic.php?f=16&t=1731

spirkaa, 2016-05-20:
I still get this error on WeMos D1 mini with MicroPython v1.8-89-gf9dc644-dirty on 2016-05-20, but can avoid it. What i do:
1. Flash  [WiFiClientBasic.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiClientBasic/WiFiClientBasic.ino) example from Arduino esp8266 with my ssid & password
2. Flash micropython WITHOUT erase_flash
3. Run repl, paste and run do_connect()

MarkR42, 2016-06-06:
Release notes from v1.5.4 of SDK are here:

http://bbs.espressif.com/viewtopic.php?f=46&t=2198

According to this, they have fixed an issue: "The ESP8266 maybe reset if connected to a special router."

This could be the same issue.

I've hacked the 1.5.4. sdk to work with micropython and I'm testing it soon with my issue.