micropython/micropython #1947: esp8266 os.listdir error 19

vfl68, 2016-03-30:
Hello
I've the following error
```
MicroPython v1.6-334-gb59f4f4 on 2016-03-30; ESP module with ESP8266
Type "help()" for more information.
>>> import os
>>> os.listdir()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 19
````

I've cleaned and rebuilded the esp-open-sdk and mp for esp8266 from github.
I don't think it's a hardware problem even with the esp board from olimex because with the kickstarter firmware v02, os.listdir() works fine.

I also tried the mpfshell, I can connect, launch repl, but a 'ls' gives a communication error with the github build.
With the kickstarter firmware v02 mpfshell and even mpfmount works fine.

deshipu, 2016-03-30:
I also managed to get this:

```
>>> import os
>>> os.listdir()
Fatal exception 4(Level1InterruptCause):
epc1=0x400005cb, epc2=0x00000000, epc3=0x00000000, excvaddr=0x2500001d, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 31664, room 16 
tail 0
chksum 0xe3
load 0x3ffe8000, len 996, room 8 
tail 12
chksum 0x5c
ho 0 tail 12 room 4
load 0x3ffe83f0, len 29800, room 12 
tail 12
chksum 0x52
csum 0x52
|l��|��rrnb��ll�b�lb쌜���b�lrldon't use rtc mem data
ll��|��rrnb����l�b�b쌜���b��lrl�l��|��rrnb��l����b�b쌜��l�p�b�lblrl��rl���bl�br|�bll���lb��n�nn��l�|�l�l��l������l`�n����b��b���bl��could not find module 'boot'

>>> 
MicroPython v1.6-327-g61fa7c8-dirty on 2016-03-30; ESP module with ESP8266
Type "help()" for more information.
>>> ets_alt_task: ets_run
q for task 2: queue: 3fff9668, get ptr: 0, put ptr: 0, qlen: 16
q for task 20: queue: 3fff05c0, get ptr: 0, put ptr: 0, qlen: 3
q for task 28: queue: 3fffa2d8, get ptr: 0, put ptr: 0, qlen: 10
q for task 31: queue: 3ffef870, get ptr: 0, put ptr: 0, qlen: 4
q for task 32: queue: 3fff0830, get ptr: 0, put ptr: 0, qlen: 34
====

>>> 
```

dhylands, 2016-03-30:
I'm going to guess that this has something to do with not passing any arguments to listdir. Does os.listdir('.') fail the same way?

trixam, 2016-03-31:
```
MicroPython v1.6-327-g61fa7c8 on 2016-03-30; ESP module with ESP8266
Type "help()" for more information.
>>> import os
>>> os.listdir('.')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 19
>>> 

Same.
```

pfalcon, 2016-03-31:
errno 19 is (http://www.virtsync.com/c-error-codes-include-errno) ENODEV      19  /* No such device */ . You didn't mount a filesystem, did you?

vfl68, 2016-03-31:
No I didn't mount a filesystem, didn't find an os.mount or pyb.mount ... can you tell me how to mount the filesystem ?

gschmottlach, 2016-04-02:
I have the same problem as vfl68 mentioned. It's not clear to me how to mount the file system on the ESP8266. Could someone offer a suggestion? It would be great if this were just an "operator" error rather than a real bug in the software.

pfalcon, 2016-04-11:
Filesystem bootstrap is now in master. Considering this bug being about segfault @deshipu got.

dpgeorge, 2016-04-14:
I found a reliable way to reproduce this bug: it only occurs on initial boot with a fresh module, and it's because a partially mounted filesystem was left registered.  Should be fixed by bcd719ea3afa9d4dd8e1d8f5168e60e5c9fa992e.

pfalcon, 2016-04-14:
@dpgeorge : Cool, thanks!

iSevenDays, 2016-06-17:
Hello everyone

I also didn't find how to mount filesystem

```
=== import os
=== os.listdir('.')
=== 
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
OSError: [Errno 19] ENODEV

```

=== import port_diag
=== 
FlashROM:
Flash ID: 1640e0 (Vendor: e0 Device: 4016)
Flash bootloader data:
Byte @2: 00
Byte @3: 40 (Flash size: 4MB Flash freq: 40MHZ)

Networking:
STA ifconfig: ('0.0.0.0', '0.0.0.0', '0.0.0.0', '208.67.222.222')
AP ifconfig: ('0.0.0.0', '0.0.0.0', '0.0.0.0', '208.67.222.222')
Free WiFi driver buffers of type:
0: 8
1: 0
2: 8
3: 4
4: 7
lwIP PCBs:
Active PCB states:
Listen PCB states:
TIME-WAIT PCB states:

Please help me

pfalcon, 2016-06-17:
Filesystem setup happens automatically on boot, and is done by: https://github.com/micropython/micropython/blob/master/esp8266/scripts/inisetup.py . You would need to follow those steps manually to see where and why it gets wrong. Also see http://forum.micropython.org/viewtopic.php?f=16&t=1689 for the information a bugreport should include.

iSevenDays, 2016-06-17:
@pfalcon thank you, here is what I have

please note that my flash size is 32m

1. Create case-sensitive dmg 
2. Install esp-open-sdk version 1.5.4
3.  make
4. export PATH
5. clone latest micro python 
6. make axtls

sevens:esp8266 seven$ esptool.py --port /dev/tty.SLAB_USBtoUART erase_flash
esptool.py v1.2-dev
Connecting...
Erasing flash (this may take a while)...
sevens:esp8266 seven$ make PORT=/dev/tty.SLAB_USBtoUART deploy
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Generating build/frozen.c
mkdir -p build/genhdr
Generating build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
mkdir -p build/build
mkdir -p build/drivers/dht
mkdir -p build/extmod
mkdir -p build/lib/fatfs
mkdir -p build/lib/fatfs/option
mkdir -p build/lib/libc
mkdir -p build/lib/libm
mkdir -p build/lib/mp-readline
mkdir -p build/lib/netutils
mkdir -p build/lib/timeutils
mkdir -p build/lib/utils
mkdir -p build/py
mkdir -p build/stmhal
CC ../py/mpstate.c
CC ../py/nlrx86.S
CC ../py/nlrx64.S
CC ../py/nlrthumb.S
CC ../py/nlrxtensa.S
CC ../py/nlrsetjmp.c
CC ../py/malloc.c
CC ../py/gc.c
CC ../py/qstr.c
CC ../py/vstr.c
CC ../py/mpprint.c
CC ../py/unicode.c
CC ../py/mpz.c
CC ../py/lexer.c
CC ../py/lexerstr.c
CC ../py/lexerunix.c
CC ../py/parse.c
CC ../py/scope.c
CC ../py/compile.c
CC ../py/emitcommon.c
CC ../py/emitbc.c
CC ../py/asmx64.c
CC ../py/emitnative.c
CC ../py/asmx86.c
CC ../py/emitnative.c
CC ../py/asmthumb.c
CC ../py/emitnative.c
CC ../py/emitinlinethumb.c
CC ../py/asmarm.c
CC ../py/emitnative.c
CC ../py/formatfloat.c
CC ../py/parsenumbase.c
[... 209 lines cut ...]
""")
        time.sleep(3)

def setup():
    check_bootsec()
    print("Performing initial setup")
    wifi()
    uos.VfsFat.mkfs(bdev)
    vfs = uos.VfsFat(bdev, "")
    with open("/boot.py", "w") as f:
        f.write("""\
# This file is executed on every boot (including wake-boot from deepsleep)
#import webrepl
#webrepl.start()
""")
    return vfs

setup()

it says 

FAT filesystem appears to be corrupted. If you had important data there, you
may want to make a flash snapshot to try to recover it. Otherwise, perform
factory reprogramming of MicroPython firmware (completely erase flash, followed
by firmware programming)

iSevenDays, 2016-06-17:
Also I modified script check boot, maybe it will help:

```
import uos
from flashbdev import bdev

try:
    if bdev:
        print("bdev is presented")
        vfs = uos.VfsFat(bdev, "")
        print(vfs)
        buf = bytearray(bdev.SEC_SIZE)
        bdev.readblocks(0, buf)

        print(bytes(buf).decode())
        print(buf)
    else:
        print("bdev NOT found")
except OSError:
    print("OS_Error")
```

```
bdev is presented
<VfsFat>
���MSDOS5.0   s� ? �         � )  !(NO NAME    FAT                                                                                                                                                                                                                                                                                                                                                                                                                                                                     U�                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
bytearray(b'\xeb\xfe\x90MSDOS5.0\x00\x10\x01\x01\x00\x01\x00\x02s\x03\xf0\x01\x00?\x00\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\x00)\x00\x00!(NO NAME    FAT     \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0 [... cut ...] 0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')

```

iSevenDays, 2016-06-17:
@pfalcon 
Okay, I have found the root of the problem

=== try:
===     if bdev:
===         print("bdev is presented")
=== 
===         print(uos.VfsFat.mkfs(bdev))
===         #print(vfs = uos.VfsFat(bdev, ""))
===         
===         
===     else:
===         print("bdev NOT found")
=== except OSError as err:
===     print("OS_Error:", err)
=== 

bdev is presented
OS_Error: **too many devices mounted**

pfalcon, 2016-06-17:
Sorry, based on the information you provided, I still don't see what can be wrong (except for universal explanation - hardware, e.g. power supply, problems.)

The dump you show is pretty valid FAT boot sector from a first look (I won't see a mis-flipped bit in it of course).   "OS_Error: too many devices mounted" is simply because you ran .mount more than once. At this point, I'd disable all boot-time processing in _boot.py, erased flash, and run .mkfs() and .mount manually to see how they behave in real-time.

iSevenDays, 2016-06-17:
@pfalcon thanks for the quick response

I will modify inisetup from frozen.c to see if there is an exception that is not printed at first launch after flash

iSevenDays, 2016-06-17:
@pfalcon unfortunately I don't know why message from inisetup "Performing initial setup" is not printing at first launch after boot

Here is my modified version

`"import uos\nimport network\nfrom flashbdev import bdev\n\ndef wifi():\n    import ubinascii\n    ap_if = network.WLAN(network.AP_IF)\n    essid = b\"MicroPython-%s\" % ubinascii.hexlify(ap_if.config(\"mac\")[-3:])\n    ap_if.config(essid=essid, authmode=network.AUTH_WPA_WPA2_PSK, password=b\"micropythoN\")\n\ndef check_bootsec():\n    buf = bytearray(bdev.SEC_SIZE)\n    bdev.readblocks(0, buf)\n    empty = True\n    for b in buf:\n        if b != 0xff:\n            empty = False\n            break\n    if empty:\n        return True\n    fs_corrupted()\n\ndef fs_corrupted():\n    import time\n    while 1:\n        print(\"\"\"\\\nFAT filesystem appears to be corrupted. If you had important data there, you\nmay want to make a flash snapshot to try to recover it. Otherwise, perform\nfactory reprogramming of MicroPython firmware (completely erase flash, followed\nby firmware programming).\n\"\"\")\n        time.sleep(3)\n\ndef setup():\n    check_bootsec()\n    print(\"Performing initial setup\")\n    wifi()\n    uos.VfsFat.mkfs(bdev)\n    vfs = None\n    try:\n        vfs = uos.VfsFat(bdev, \"\")\n        with open(\"/boot.py\", \"w\") as f:\n            f.write(\"\"\"\\\n# This file is executed on every boot (including wake-boot from deepsleep)\n#import webrepl\n#webrepl.start()\n\"\"\")\n    except Exception as err:\n        print(\"Init error fat\", err)\n    return vfs\n\0"`

pfalcon, 2016-06-17:
Neither I, that's why I wrote:

> At this point, I'd disable all boot-time processing in _boot.py, erased flash, and run .mkfs() and .mount manually to see how they behave in real-time.

What exact module model do you use and how do you power it?

iSevenDays, 2016-06-17:
@pfalcon thank you, I tried

```
=== import uos
=== import network
=== from flashbdev import bdev
=== 
=== def check_bootsec():
===     buf = bytearray(bdev.SEC_SIZE)
===     bdev.readblocks(0, buf)
===     empty = True
===     for b in buf:
===         if b != 0xff:
===             empty = False
===             break
===     if empty:
===         return True
===     fs_corrupted()
=== 
=== def fs_corrupted():
===     import time
===     while 1:
===         print("""
=== FAT filesystem appears to be corrupted. If you had important data there, you
=== may want to make a flash snapshot to try to recover it. Otherwise, perform
=== factory reprogramming of MicroPython firmware (completely erase flash, followed
=== by firmware programming).
=== """)
===         time.sleep(3)
=== 
=== def setup():
===     print("Performing initial setup")
===     #wifi()
===     try:
===         print("Executing mkfs")
===         uos.VfsFat.mkfs(bdev)
===     except OSError as err:
===         print("OSError error fat", err)
===     except Exception as err:
===         print("Init error fat", err)
===     vfs = None
===     try:
===         print("vfs = uos.VfsFat(bdev, )")
===         vfs = uos.VfsFat(bdev, "")
===         with open("/boot.py", "w") as f:
===             print("/boot.py opened, do noting")
===     except OSError as err:
===         print("OSError2 error fat", err)
===     except Exception as err:
===         print("Init2 error fat", err)
===     return vfs
=== 
=== setup()
=== 

```

with check boot uncommented I always have fs_corrupted() executed

1. esptool.py --port /dev/tty.SLAB_USBtoUART flash_id
Manufacturer: e0
Device: 4016

2.  I have esp8266 12, I power it from power source ams1117 3.3 from 1A wall charger
I have 2 bypass capacitors, 10k pullup chip enable, 10k pulldown gpio15

I tried baud rate 9600 - no difference,  also seems that --flash_mode dio makes no difference

thanks

iSevenDays, 2016-06-18:
@pfalcon sorry, I was thinking that frozen.c is written by a human, not auto-generated

So then I edited real _boot.py and inisetup.py files

I modified simple script

```
import uos
import network
from flashbdev import bdev

def check_bootsec():
    print("Executing check_bootsec")
    buf = bytearray(bdev.SEC_SIZE)
    bdev.readblocks(0, buf)
    empty = True
    for b in buf:
        if b != 0xff:
            print("check_bootsec is empty FALSE")
            empty = False
            break
    if empty:
        print("check_bootsec is empty TRUE")
        return True
    fs_corrupted()

def fs_corrupted():
    import time
    while 1:
        print("""
FAT filesystem appears to be corrupted. If you had important data there, you
may want to make a flash snapshot to try to recover it. Otherwise, perform
factory reprogramming of MicroPython firmware (completely erase flash, followed
by firmware programming).
""")
        time.sleep(3)

def setup():
    print("Performing initial setup")
    check_bootsec()
    #wifi()
    try:
        print("Executing mkfs")
        uos.VfsFat.mkfs(bdev)
    except OSError as err:
        print("OSError error fat", err)
    except Exception as err:
        print("Init error fat", err)
    vfs = None
    try:
        print("vfs = uos.VfsFat(bdev, )")
        vfs = uos.VfsFat(bdev, "")
        with open("/boot.py", "w") as f:
            print("/boot.py opened, do noting")
    except OSError as err:
        print("OSError2 error fat", err)
    except Exception as err:
        print("Init2 error fat", err)
    print("Setup done")
    return vfs

setup()
```

Output:

```
Performing initial setup
Executing check_bootsec
check_bootsec is empty TRUE
Executing mkfs
vfs = uos.VfsFat(bdev, )
/boot.py opened, do noting
Setup done
<VfsFat>

=== import os
=== os.listdir('.')
=== 
['boot.py']
```

After restarting the board I have the same correct result, I don't know why I can't flash second board, maybe it has corrupted flash.

Then I reverted all my changes made to .py files and flashed first working board, everything was okay, but boot.py is missing, strange, I thought it should be created automatically.

```
=== import os
=== os.listdir('.')
=== 
[]

```

iSevenDays, 2016-06-18:
And then I checked that I am unable to write files, I have exception

```
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== with open("/boot.py", "w") as f:
===     f.write("""\
=== # This file is executed on every boot (including wake-boot from deepsleep)
=== #import webrepl
=== #webrepl.start()
=== """)
=== 
108

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30792, room 16 
tail 8
chksum 0x05
load 0x3ffe8000, len 1044, room 0 
tail 4
chksum 0x44
load 0x3ffe8420, len 3116, room 4 
tail 8
chksum 0x19
csum 0x19
lܞ|��r{ob��l��b�lb䌜������c�$slll��|����rnc���l܎B�#�ܜ�{#��cd`� l��|��2r{Nc��$��c�c䌜��$��b�l#l��r�d����{l��{�l�l�l ��{�l��� $l r$���n��c#l�c{|�bp��cb#��n�on� ll��l� �$��ld���l��$`�N� ��b#l䏌b��b#l�#p{ls$r�N�� ���n�߀�c{#��"�b�����pcl��lo�prl���could not open file 'boot.py' for reading
could not open file 'main.py' for reading

#4 ets_task(40100350, 3, 3fff6360, 4)
MicroPython v1.8.1-58-g7053621-dirty on 2016-06-18; ESP module with ESP8266
Type "help()" for more information.

```

And after I think I corrupted file system

```
os.listdir('.')
=== 
['b\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00']

```

pfalcon, 2016-06-18:
Well, then indeed hardware problem is the only explanation. Other folks, e.g. @deshipu who faced similar problem can to the same conclusion IIRC. I personally somehow doubt it's corrupted flash (maybe because I didn't face such cases in real life), I'd suspect power supply problems (high dropout, high ripple, etc.)

iSevenDays, 2016-06-24:
@pfalcon 

Hello, I've bought another 2 boards and I tried to flash them the usual way. It didn't work for me.

Then I tried my script for creating fat fs and also I disabled default boot, inisetup

On 2 boards I've got

```
MicroPython v1.8.1-72-gad22947-dirty on 2016-06-25; ESP module with ESP8266
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import uos
=== import network
=== from flashbdev import bdev
=== 
=== def check_bootsec():
===     print("Executing check_bootsec")
===     buf = bytearray(bdev.SEC_SIZE)
===     bdev.readblocks(0, buf)
===     empty = True
===     for b in buf:
===         if b != 0xff:
===             print("check_bootsec is empty FALSE")
===             empty = False
===             break
===     if empty:
===         print("check_bootsec is empty TRUE")
===         return True
===     fs_corrupted()
=== 
=== def fs_corrupted():
===     import time
===     while 1:
===         print("""
=== FAT filesystem appears to be corrupted. If you had important data there, you
=== may want to make a flash snapshot to try to recover it. Otherwise, perform
=== factory reprogramming of MicroPython firmware (completely erase flash, followed
=== by firmware programming).
=== """)
===         time.sleep(3)
=== 
=== def setup():
===     print("Performing initial setup")
===     #check_bootsec()
===     #wifi()
===     try:
===         print("Executing mkfs")
===         print(uos.VfsFat.mkfs(bdev))
===     except OSError as err:
===         print("OSError error fat", err)
===     except Exception as err:
===         print("Init error fat", err)
===     vfs = None
===     try:
===         print("vfs = uos.VfsFat(bdev, )")
===         vfs = uos.VfsFat(bdev, "")
===         with open("/boot.py", "w") as f:
===             print("/boot.py opened, do noting")
===     except OSError as err:
===         print("OSError2 error fat", err)
===     except Exception as err:
===         print("Init2 error fat", err)
===     print("Setup done")
===     return vfs
=== 
=== setup()
=== 
Performing initial setup
Executing mkfs

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 30792, room 16 
tail 8
chksum 0x5b
load 0x3ffe8000, len 1044, room 0 
tail 4
chksum 0x6a
load 0x3ffe8420, len 3116, room 4 
tail 8
chksum 0x37
csum 0x37
��$ܟ|��r{ob��l��#�dc������c�#$`dl�����r{ob���l܎��"�#���rB��bl �dܾ�����rNc�����$"�#쌜��d��B�$b$��r�l�l� �$`��{�l�$�l �o�$���dl`rl��o���#l�br|�`�2��c"c��o�2No� dl��l� �l�� $l����l��$`�o� ���c"l䎌b���bd�Bps$r$r�o�� � ��oܞ2Brb�������$�����lcl��$o�bdev is True
OSERROR:  can't mount
inisetup.setup()
could not open file 'boot.py' for reading
could not open file 'main.py' for reading

#4 ets_task(40100350, 3, 3fff6360, 4)
MicroPython v1.8.1-72-gad22947-dirty on 2016-06-25; ESP module with ESP8266
Type "help()" for more information.

```

Seems that I am unable to debug uos.VfsFat.mkfs(bdev) within Python

Is there any way to enable debug information logging for VfsFat?

Because I have 3 boards, 1 of them flashed successfully, other 2 not. I think that it would be very strange if it is memory corruption problem on 2 boards.

iSevenDays, 2016-06-25:
@pfalcon 

I figured out how to print messages to python console from c code and I have interesting results.

I found the line that crashes my board.

The problem line is in **ff.c**
if (disk_write(pdrv, tbl, wsect++, 1) != RES_OK) {

Here is the full output log.

```
PORT OPEN 115200
 {l $ܞ| �l�|�l�c|����{�c�c��oN� $no���bp�#$`rlp�n��lbN�|���#�|~�o�l�� l �2Nol n{���Nrl`p�N�s�����Bo�|$��b��oN�$`�oN�dor�ےn2$� p�N�s�������Bo�|�l$��p��no��l`�nod`or���n�b� r��N�"�`��$ܞ|����rn#���l��#��c䌜�����#�#$`ll��|�����rNb��l܎c�p쌜��{b��cd`�l��|��{rnc�����lc�B쌜��l��b�$b$��r�l�$��l`��{�$�l� l ��r�l���$l`rl��sl���bl�br|�`����bc#�p�non� $l��l쏌�$��$l���$��d �n� ���bl䏌#��bcd�#prdrl{�N�� ���o����brb�pb�bl�l���l�l{$��lN�p{l�l��|��{Z�bdev is  <FlashBdev object at 3ffef840>
bdev is True
Executing fatfs_mount_mkfs
fatfs_mount_mkfs parse args
fatfs_mount_mkfs get the mount point
fatfs_mount_mkfs create new object
fatfs_mount_mkfs load block protocol methods
fatfs_mount_mkfs device supports new block protocol
fatfs_mount_mkfs Register the vfs object so that it can be found by the FatFS driver using ff_get_ldnumber
ioctl(1, 0)
ioctl(5, 0)
readblocks(0, 3ffefae0(4096))
OSERROR:  can't mount
inisetup.setup()
could not open file 'boot.py' for reading
could not open file 'main.py' for reading

#4 ets_task(40100350, 3, 3fff6360, 4)
MicroPython v1.8.1-72-gad22947-dirty on 2016-06-25; ESP module with ESP8266
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import uos
=== import network
=== from flashbdev import bdev
=== 
=== def check_bootsec():
===     print("Executing check_bootsec")
===     buf = bytearray(bdev.SEC_SIZE)
===     bdev.readblocks(0, buf)
===     empty = True
===     for b in buf:
===         if b != 0xff:
===             print("check_bootsec is empty FALSE")
===             empty = False
===             break
===     if empty:
===         print("check_bootsec is empty TRUE")
===         return True
===     fs_corrupted()
=== 
=== def fs_corrupted():
===     import time
===     while 1:
===         print("""
=== FAT filesystem appears to be corrupted. If you had important data there, you
=== may want to make a flash snapshot to try to recover it. Otherwise, perform
=== factory reprogramming of MicroPython firmware (completely erase flash, followed
=== by firmware programming).
=== """)
===         time.sleep(3)
=== 
=== def setup():
===     print("Performing initial setup")
===     #check_bootsec()
===     #wifi()
===     try:
===         print("Executing mkfs")
===         print(uos.VfsFat.mkfs(bdev))
===     except OSError as err:
===         print("OSError error fat", err)
===     except Exception as err:
===         print("Init error fat", err)
===     vfs = None
===     try:
===         print("vfs = uos.VfsFat(bdev, )")
===         vfs = uos.VfsFat(bdev, "")
===         with open("/boot.py", "w") as f:
===             print("/boot.py opened, do noting")
===     except OSError as err:
===         print("OSError2 error fat", err)
===     except Exception as err:
===         print("Init2 error fat", err)
===     print("Setup done")
===     return vfs
=== 
=== setup()
=== 
Performing initial setup
Executing mkfs
Executing fat_vfs_mkfs
Executing fatfs_mount_mkfs
fatfs_mount_mkfs parse args
fatfs_mount_mkfs get the mount point
fatfs_mount_mkfs create new object
fatfs_mount_mkfs load block protocol methods
fatfs_mount_mkfs device supports new block protocol
fatfs_mount_mkfs Register the vfs object so that it can be found by the FatFS driver using ff_get_ldnumber
Executing f_mkfs
f_mkfs Check mounted drive and clear work area
ioctl(1, 0)
ioctl(5, 0)
f_mkfs Create a partition in this function
ioctl(4, 0)
f_mkfs Align data start sector to erase block boundary (for flash memory media)
f_mkfs Determine system ID in the partition table
f_mkfs FAT12
f_mkfs No partition table (SFD)
f_mkfs Create BPB in the VBR
f_mkfs executed: mem_set(tbl, 0, SS(fs));
f_mkfs executed: mem_cpy(tbl, ��� MSDOS5.0, 11)
f_mkfs executed: ST_WORD(tbl + BPB_BytsPerSec, i);
f_mkfs executed: ST_WORD(tbl + BPB_RsvdSecCnt, n_rsv)
f_mkfs executed: ST_WORD(tbl + BPB_RootEntCnt, i);
f_mkfs executed: ST_WORD(tbl + BPB_TotSec16, n_vol);
f_mkfs executed: ST_WORD(tbl + BPB_SecPerTrk, 63)
f_mkfs executed: ST_WORD(tbl + BPB_NumHeads, 255);
f_mkfs executed: mem_cpy(tbl + BS_VolLab, NO NAME     FAT     , 19)
f_mkfs executed: ST_WORD(tbl + BS_55AA, 0xAA55)
writeblocks(0, 3fff0090(4096))
f_mkfs executed: disk_write(pdrv, tbl, b_vol, 1) != RES_OK
f_mkfs  Initialize FAT area
f_mkfs Initialize each FAT copy, i = 0
f_mkfs trying to perform mem_set(tbl, 0, SS(fs));
f_mkfs mem_set(tbl, 0, SS(fs)); success
f_mkfs trying to perform if (disk_write(pdrv, tbl, wsect++, 1) != RES_OK) {
writeblocks(1, 3fff0270(4096))
f_mkfs disk_write(pdrv, tbl, wsect++, 1) success
f_mkfs trying to perform mem_set(tbl, 0, SS(fs));
f_mkfs mem_set(tbl, 0, SS(fs)); success
f_mkfs Initialize root directory
f_mkfs fmt == FS_FAT32 1
f_mkfs trying to execute disk_write(pdrv, tbl, wsect++, 1) != RES_OK)
with arguments: pdrv: 0, tbl: 1073688144, wsect: 3writeblocks(2, 3fff02d0(4096))

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 30792, room 16 
tail 8
chksum 0xa3
load 0x3ffe8000, len 1044, room 0 
tail 4
chksum 0x9a
load 0x3ffe8420, len 3116, room 4 
tail 8
chksum 0xdb
csum 0xdb

```

[diskio_ff.zip](https://github.com/micropython/micropython/files/332833/diskio_ff.zip)
[fusermount_vfsfat.zip](https://github.com/micropython/micropython/files/332832/fusermount_vfsfat.zip)

iSevenDays, 2016-06-25:
Hello @pfalcon 

I would like to thank you for the thoughts about power supply.

vfs = uos.VfsFat(bdev, )
Executing fatfs_mount_mkfs
fatfs_mount_mkfs parse args
fatfs_mount_mkfs get the mount point
fatfs_mount_mkfs create new object
fatfs_mount_mkfs load block protocol methods
fatfs_mount_mkfs device supports new block protocol
fatfs_mount_mkfs Register the vfs object so that it can be found by the FatFS driver using ff_get_ldnumber
ioctl(1, 0)
ioctl(5, 0)
readblocks(0, 3fff04f0(4096))
readblocks(2, 3fff0540(4096))
/boot.py opened, do noting
writeblocks(2, 3fff05d0(4096))
ioctl(3, 0)
Setup done
<VfsFat>
>>> chg_A3:0

**Q**: Where was the problem?
**A**: In power supply. I have macbook that conducts current through the body.

Here is some things that I face like "pinching" and how to resolve it http://garethhayes.net/apple-power-plug-not-earthed/ , but I didn't think it was the big problem for the ESP8266 12

I don't have the "earth" pin, so to get the ESP8266 flashed correctly I disconnected power supply and moved my FTDI chip as far as possible from my netbook.

Also I have added a lot of capacitors, but only after I disconnected laptop power supply I was able to launch fat filesystem.

@pfalcon , could you please add this to FAQ?

I think that I'm not the only one with this problem, but maybe others didn't spend 2 weeks+ of trying and resolving all the problems.

Thanks

pfalcon, 2016-07-22:
Information added to the FAQ at http://forum.micropython.org/viewtopic.php?f=16&t=1908&p=10759 . Closing.

closed 2016-07-22 by pfalcon

peterhinch, 2016-07-24:
@pfalcon Computers without an earth connection present a risk of physical damage to external electronics such as ESP8266 or Pyboards. An earth should be in place before plugging in any exposed PCB's. I'd be willing to write something on the subject, but it's unclear where in the docs it should go as it applies to all MicroPython targets. Reading this issue I was astounded that such machines exist - but then I considered devices like a Raspberry Pi powered from a USB wall-wart of unknown provenance.

wavesailor, 2017-02-14:
I had the same issue as originally mentioned. I'm using an ESP8266 board (https://www.adafruit.com/product/2471). Seemed like the file system was corrupt or not loading. 

What I did to fix it was to enable the AP
`ap = network.WLAN(network.AP_IF) # create access-point interface
ap.active(True)         # activate the interface
ap.config(essid='ESP-AP') # set the ESSID of the access point`

After rebooting I could then access the file system with
`>>> import os
>>> os.listdir()`

Hope  this helps someone else as it took me ages to solve it.