micropython/micropython #1951: UNIX port: add option to start repl after a script is executed.

prusnak, 2016-03-31:
STMHAL port works like this:
1) run the contents of boot.py
2) run the contents of main.py
3) start REPL

If any of steps 1) or 2) contain an infinite loop, the following steps are not reached.

UNIX port has currently 2 modes of operation:
a) start REPL (when no script filename is provided)
b) run script and exit (when a script filename is provided)

I want to emulate exactly the STMHAL behaviour using UNIX port, but currently it is not possible.

Suggestion 1: add `-r` argument to UNIX micropython, which instructs it not to exit after the script is executed, but start REPL instead.

Suggestion 2: add possibility to specify more than one script on command line and they will be executed one after each other.

When both of these are implemented, the STMHAL behaviour would be achieved by running:

```
./micropython -r boot.py main.py
```

Suggestion #2 is just a nice to have feature, because the same behaviour could be emulated by using:

```
cat boot.py main.py | ./micropython
```

However, suggestion #1 can't be AFAIK achieved by any means at the moment.

prusnak, 2016-03-31:
I just realized there is an undocumented `-m module` switch and `./micropython -m main` behaves exactly like I want (starts REPL after module is finished). However, running more than one module is not possible and `cat` trick is not possible.

pfalcon, 2016-03-31:
Run `python3 --help` and look where you see -r switch or ability to run multiple scripts. So, it's not unix port what should be brought in correspondence with stmhal port, but stmhal port in correspondence with the Python language (things like https://github.com/micropython/micropython/issues/1741 , etc.)

> However, suggestion 1 can't be AFAIK achieved by any means at the moment.

You're welcome to design and implement (subset) of "code" module in a way which doesn't cause bloat in case it's not enabled: https://docs.python.org/3/library/code.html#code.interact

pfalcon, 2016-03-31:
> I just realized there is an undocumented -m module switch and ./micropython -m main behaves exactly like I want (starts REPL after module is finished).

-m is the standard Python switch, it's not intended to work as you describe, and it doesn't. If you see otherwise, please report that as a bug, it will be fixed.

> However, running more than one module is not possible and cat trick is not possible.

Obviously, because whatever comes after script name, or -m module are command line args to script.

prusnak, 2016-03-31:
Closing then.

closed 2016-03-31 by prusnak

ncoghlan, 2016-04-01:
Does the UNIX port of MicroPython implement the "-i" switch? That stands for "inspect" (with corresponding environment variable PYTHONINSPECT) and drops you into the REPL with a fully populated __main__, even if another command has been executed first: https://docs.python.org/3/using/cmdline.html#cmdoption-i

prusnak, 2016-04-01:
Implemented -i option in https://github.com/micropython/micropython/pull/1952