micropython/micropython #1954: "make" esp8266 not working on Ubuntu 14.04

Adixy, 2016-04-01:
Greetings
I have installed esp open sdk and everythin that comes with it, including esptool
But when I cd esp8266 and start >>>make, this shows:
Makefile:1: ../py/mkenv.mk: No such file or directory
Makefile:7: ../py/py.mk: No such file or directory
Makefile:194: ../py/mkrules.mk: No such file or directory
make: *** No rule to make target `../py/mkrules.mk'.  Stop.
Can you please help me, does any1 know hot to fix it? Am I doing something wrong?
I have bought Huzzah esp8266 and spent more than20 hours just trying to put micropython on it, still not succeding.
Thank you :D

vfl68, 2016-04-02:
Hi,
I'm sorry I can't reproduce your error.
Did you clone the whole micropython project or did you only get the files from the micropython/esp8266 directory ?

Adixy, 2016-04-02:
Hi, 
oh no worries about that:)
I cloned only the files from the esp8266

Adixy, 2016-04-02:
I just cloned the whole micropython-master and it works
Thank you very much :D

closed 2016-04-02 by pfalcon