micropython/micropython #1957: ESP8266 boot loops

Adixy, 2016-04-02:
whenever I boot-up ESP or restart it (the button on it),
he enters this loop and it stays in it for way too long
Isn't there a watchdog to stop it? or some interrupt or whatever..

#4 ets_task(40100264, 3, 3fff95b8, 4)
MicroPython v1.6-340-ga0cb4ed on 2016-04-02; ESP module with ESP8266
Type "help()" for more information.
>>> Fatal exception 0(IllegalInstructionCause):
epc1=0x402151c0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x40100000, len 31840, room 16 
tail 0
chksum 0x8b
load 0x3ffe8000, len 1000, room 8 
tail 0
chksum 0x60
load 0x3ffe83f0, len 29880, room 8 
tail 0
chksum 0x80
csum 0x80
$śž|˙�r{ocŹŚdl$‡cě$#ěŚśśăě#Çdsdldśž|˙�;socŽÄŚŚä#ěŚcäÄÜśścě‡lrl„lśź|ţ�{;gc‡„$Śś€ä�c„bě„śÜśŚÇx€#älbdrl‡�;d„źăŚcdŚc{|ŚcdlŹ‡źlcäň'ç€noăÄdŕ|„$„l„‡däçcould not find module 'boot'

#4 ets_task(40100264, 3, 3fff95b8, 4)
MicroPython v1.6-340-ga0cb4ed on 2016-04-02; ESP module with ESP8266
Type "help()" for more information.
>>> ty@@t€DPD I‡`-=Mh]KfEŐ˙‘HVůrÄˇbaRb€f`U PPôW·Ţ˙r.Sf'!	‘Ç`-=MĹŕ˙Vö‘baRb€f`U PPô(LoadStoreAlignmentCause):
epc1=0x4022b25f, epc2=0x00000000, epc3=0x00000000, excvaddr=0x402291a2, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x40100000, len 31840, room 16 
tail 0
chksum 0x8b
load 0x3ffe8000, len 1000, room 8 
tail 0
chksum 0x60
load 0x3ffe83f0, len 29880, room 8 
tail 0
chksum 0x80
csum 0x80
$Üź<˙�s{g#ŹŚdd$Çcä$bě„śÜăäbŽds$don't use rtc mem data
llÜź<˙�s{o#ŹŚ„Śäcě„cäŚśśÜcäÇdsdÄdśß|˙‚;sncÇÄl„śŔä‚#ÄcěÄśśśŚŹ8€cědc$s$ŹĂslŚžă„cl„cr|„cldÇŹźlcěóoďŔo'ăŚlŕ|ÄlŚ$ŚŽlěçcould not find module 'boot'

#4 ets_task(40100264, 3, 3fff95b8, 4)
MicroPython v1.6-340-ga0cb4ed on 2016-04-02; ESP module with ESP8266
Type "help()" for more information.
>>> Fatal exception 28(LoadProhibitedCause):
epc1=0x402141e3, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000e0, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x40100000, len 31840, room 16 
tail 0
chksum 0x8b
load 0x3ffe8000, len 1000, room 8 
tail 0
chksum 0x60
load 0x3ffe83f0, len 29880, room 8 
tail 0
chksum 0x80
csum 0x80
$Üź<˙�s{g#ŹŚdd$Çcä$bě„śÜăäbŽds$don't use rtc mem data
dlśź|ţ�{;gb‡„Ä„ěcäŚbě„śśś#äŹ${lŚlÜź|˙�r{ocŹŚdÄÜ€ě�cŚcäŚśśś„‡pŔcä$clsdŽ�slŚßăŚblŚb;|„cll‡Žźlběóoî€ogăŚlŕ|ÄlÄdŚÇ$ěćcould not find module 'boot'

#4 ets_task(40100264, 3, 3fff95b8, 4)
MicroPython v1.6-340-ga0cb4ed on 2016-04-02; ESP module with ESP8266
Type "help()" for more information.
>>> Fatal exception 28(LoadProhibitedCause):
epc1=0x402141e3, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000fc, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x40100000, len 31840, room 16 
tail 0
chksum 0x8b
load 0x3ffe8000, len 1000, room 8 
tail 0
chksum 0x60
load 0x3ffe83f0, len 29880, room 8 
tail 0
chksum 0x80
csum 0x80
lśź|˙�s;gb‡„$dlŽcälcě„śśăäcŹ${ldon't use rtc mem data
ddśź|ţĂ{;'c‡„ŚÄěcäŚcě„śÜśběŹl{lŚlÜź<˙�s{o#ŹŚdŚÜ€ě�cŚ#äŚśśÜ„‡p€cälclrl‡�;d„źăŚcdŚc{|ŚcdlŹ‡źlcäň'ç€noăÄdŕ|„$„l„‡däçcould not find module 'boot'

#4 ets_task(40100264, 3, 3fff95b8, 4)
MicroPython v1.6-340-ga0cb4ed on 2016-04-02; ESP module with ESP8266
Type "help()" for more information.
>>> Fatal exception 28(LoadProhibitedCause):
epc1=0x402141e3, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000e0, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x40100000, len 31840, room 16 
tail 0
chksum 0x8b
load 0x3ffe8000, len 1000, room 8 
tail 0
chksum 0x60
load 0x3ffe83f0, len 29880, room 8 
tail 0
chksum 0x80
csum 0x80
dśß|˙Ă{sncŽÄlldŹbědcäŚÜśâěc‡lrldlśź|˙�s;gb‡„Ä„ěcäÄběŚśśś#äŽ${$Ś$Üź|˙�rsocŹŚdÄś€ě�bŚcäŚÜśś„ŹpŔcä$cl{$Ź‚slŚźă„cl„cs<„#ldÇŹźdcěűoďŔg'ă„lŕ|ŚlŚ$ŚŹlěçcould not find module 'boot'

#4 ets_task(40100264, 3, 3fff95b8, 4)
MicroPython v1.6-340-ga0cb4ed on 2016-04-02; ESP module with ESP8266
Type "help()" for more information.
>>> ty@@t€DPD I‡`-=Mh]KfEŐ˙‘HVůrÄˇbaRb€f`U PPôW·Ţ˙r.Sf'!	‘Ç`-=MĹŕ˙Vö‘baRb€f`U PPô(LoadProhibitedCause):
epc1=0x402141e3, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000e3, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x40100000, len 31840, room 16 
tail 0
chksum 0x8b
load 0x3ffe8000, len 1000, room 8 
tail 0
chksum 0x60
load 0x3ffe83f0, len 29880, room 8 
tail 0
chksum 0x80
csum 0x80
lÜź<˙�s{g#‡Ś$dlÇcälbě„śÜăäbŽd{$don't use rtc mem data
$dśß|˙Ă{rncÇÄŚÄě#äŚcě„śśśběŹl;l„lśź|˙�s{gb‡Ś$Śś€ě�c„#ěŚśśśÄ‡x€cäl#lsdŽĂslŚžăŚblŚbr|„cll‡Žźlběóoî€ogăŚlŕ|ÄlŚdŚŽ$ěćcould not find module 'boot'

#4 ets_task(40100264, 3, 3fff95b8, 4)
MicroPython v1.6-340-ga0cb4ed on 2016-04-02; ESP module with ESP8266
Type "help()" for more information.
>>> Fatal exception 28(LoadProhibitedCause):
epc1=0x402141e3, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000e0, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x40100000, len 31840, room 16 
tail 0
chksum 0x8b
load 0x3ffe8000, len 1000, room 8 
tail 0
chksum 0x60
load 0x3ffe83f0, len 29880, room 8 
tail 0
chksum 0x80
csum 0x80
dśß|˙‚;socŹŚdld‡cědcäŚśśăěc‡lsddon't use rtc mem data
llśź|˙�s{g#‡Ś„„äcěÄ#äŚśśÜcäŽds$Ś$śž|˙�;socŽŚl„ś€ä‚bŚcäÄÜśś„Źp€cědcl{$Ź‚slŚźă„cl„cs<„#ldÇŹźdcěűoď€g'ă„lŕ<ŚlŚ$ŚŹläçcould not find module 'boot'

#4 ets_task(40100264, 3, 3fff95b8, 4)
MicroPython v1.6-340-ga0cb4ed on 2016-04-02; ESP module with ESP8266
Type "help()" for more information.
>>> Fatal exception 0(IllegalInstructionCause):
epc1=0x4021410�, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000

Adixy, 2016-04-02:
It is because he was trying to connect to my WiFi every time it was booting up.
When I managed to:
    import network
    network.WLAN().disconnect()  (I did it with the ESPlorer, this would be impossible with putty)
while the pause between rebooting, which lasts about 300ms, it finally stabilized
Please fix this somehow, change the boot routine or smth

dpgeorge, 2016-04-04:
> It is because he was trying to connect to my WiFi every time it was booting up.

So connecting to the wifi was causing the fault and then rebooting?  Then that's a bug that should be fixed (boot routine would stay the same).  Please try again with the latest code to see if it is fixed or not.

Adixy, 2016-04-04:
The same thing happened, but this time it showed (after a few reboots) :

#4 ets_task(40100264, 3, 3fff4668, 4)
MicroPython v1.6-362-g069654f on 2016-04-04; ESP module with ESP8266
Type "help()" for more information.
>>> scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt 
chg_B1:-40

connected with OptiDSL-F8BB, channel 1
dhcp client start...
Fatal exception 0(IllegalInstructepc1=0x4021bd20, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40216b67, depc=0x00000000

and it stopped, actually freezed. When I restarted it, it finally connected to the WiFi. And I doubt that it is because my signal is week. The net modem for wifi is 1 meter from me so I have extremely strong signal

#4 ets_task(40100264, 3, 3fff4668, 4)
MicroPython v1.6-362-g069654f on 2016-04-04; ESP module with ESP8266
Type "help()" for more information.
>>> scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt 
chg_B1:-40

connected with OptiDSL-F8BB, channel 1
dhcp client start...
ip:192.168.1.9,mask:255.255.255.0,gw:192.168.1.1
pm open,type:2 0

Adixy, 2016-04-05:
load 0x40100000, len 32580, room 16 
tail 4
chksum 0xa7
load 0x3ffe8000, len 1000, room 4 
tail 4
chksum 0x23
load 0x3ffe83f0, len 360, room 4 
tail 4
chksum 0x53
csum 0x53
csum err
ets_main.c 

Now this thing pops out a few times when I turn the power on, and it just crashes. It takes me about 30 mins just to boot it up and enter the REPL mode
Maybe it is just me doing something that shouldn't be done, even though I don't know what would that be.
I use the pyboard's 3v3 as power source and CH340 as usb-ttl, everyhting on win10.
But the firmware thing is done on Linux
I hope this can be at least a little helpful to fix the bugs :D

dpgeorge, 2016-04-06:
> I use the pyboard's 3v3 as power source

Try something else.  ESP8266 requires peaks of 500mA and pyboard's LDO can supply only 300mA.

Adixy, 2016-04-06:
With a 9v battery and LM7805, together with pyb's 3v3, it boots as it should :)
Thank you, Sir, for the advice
I never expected it to "eat" that much power

closed 2016-04-07 by dpgeorge