micropython/micropython #1961: Need a quick-ref for the PYBLite board

dhylands, 2016-04-03:
It turns out that some of the pin functionailty got moved around between the lite and the regular pyboard.

For example, X1-4 on the pyboard are PA0-3, on the pyblite, its PA2, PA3, PA0, PA1

This means that some of the timers have moved around (so TIM5_CH1 is now on X3 on the pyblite where its on X1 on the pyboard.

coylen, 2016-04-06:
Its a little off the beaten path, but there is a pylite quickref in the store:
[(https://micropython.org/static/store/img/products/PYBLITEv1_0-AC-D.jpg)]

peterhinch, 2016-04-07:
I think the official Pyboard docs should detail any Lite differences. I have a PR in the pipeline (#1931) which amends the UART doc to include pblite variant pinouts. If this approach is accepted I could extend this to other class docs.

dpgeorge, 2016-08-29:
I added links to the pinouts for the lite board (and pybv1.1) on the quickref page.  I'd rather not create a separate quickref for each pyboard variant.  They are similar enough that one quickref should suffice.

dpgeorge, 2018-12-06:
A while ago in 2b7075741131c457ed0bd146cd5aecddddd5f7d2 the Servo class was updated so that X1-X4 worked correctly on PYBLITE.  Otherwise, as mentioned above, having separate pinouts for PYBLITE (which are linked from the pyboard quickref) should be enough.

closed 2018-12-06 by dpgeorge