micropython/micropython #1962: mp_fun_kw_t should take const mp_map_t* ?

pfalcon, 2016-04-03:
Is there reason we have:

~~~
typedef mp_obj_t (*mp_fun_kw_t)(size_t n, const mp_obj_t *, mp_map_t *);
~~~

Normal args are const, but kwargs map is suddenly not. This causes stuff like `(mp_map_t*)&mp_const_empty_map`.

dhylands, 2016-04-03:
I've seen lots of python code which modifies the kwargs dict that is passed in, typically when inheritance is involved.

For native code, I think that the kwargs are typically just for input, since the native stuff typically isn't involved in inheritance, so just off the top of my head I think that a change like this is reasonable.

dpgeorge, 2016-04-04:
I once tried changing the type declaration to `const mp_map_t*` but it introduced even more casting.  The problem is that you then can't call `mp_map_lookup` without casting away the const (because the lookup function is generic and can also add items, hence modify the map).  Making `mp_map_lookup` take a const map argument doesn't seem right because it may modify the map.

pfalcon, 2016-04-04:
I see, that answers it.

closed 2016-04-04 by pfalcon

pfalcon, 2016-04-04:
Actually, 069654f2be1ae370cf32a4e711f252a74326cc56

dpgeorge, 2016-04-04:
A solution would be to make a macro function like `mp_map_lookup_const(const mp_map_t *, mp_obj_t key)` that just casts away the const and calls `mp_map_lookup`.