micropython/micropython #1963: Should stream (etc.) protocol methods take object ptr, instead of mp_obj_t ?

pfalcon, 2016-04-03:
@dpgeorge, subj

dpgeorge, 2016-04-04:
Would this improve anything?  At the moment it seems that the code in stream.c does not care about the stream object (self) and it could indeed be a non-concrete object (eg an interned str) and still would work.

pfalcon, 2016-04-09:
> Would this improve anything?

Of course it improves performance. Currently, client code needs to do MP_OBJ_TO_PTR() to extract stream_p, then virtual method needs to do it again to access its object data. It could be passed pointer to object directly and save the double-work. For upcoming pin protocol this double-work is visible performance degradation (e.g. -50KHz of bitbang performance).

jonnor, 2024-07-27:
This issue has been open for 8 years, without any further comments or work. Can it be closed?

robert-hh, 2024-07-27:
Outdated. Please re-open it if needed,.

closed 2024-07-27 by robert-hh