micropython/micropython #1964: esp8266 main.py does not run on boot

ceremcem, 2016-04-04:
Hi, 

I followed [this](https://micropython.org/doc/tut-script) documentation to load a script that I want it to be run on power up of esp8266 but there is no success yet. 

Is there anyway to debug this issue (such as dumping main.py contents from REPL console)?

hoh, 2016-04-04:
The email linking to ESP8266 firmware v02 tells us to use `boot.py`, have you tried using that filename ?
> - Filesystem support is now enabled by default (and there is no way to disable it!).  Your ESP8266 device must have at least 1MB of flash for this to work, so please don't use the firmware if it doesn't.  If you create a file "boot.py" in the filesystem then this "boot.py" file will be executed on start-up.

hoh, 2016-04-04:
Related: https://github.com/micropython/micropython/pull/1936

ceremcem, 2016-04-04:
Because of [this](https://github.com/micropython/micropython/issues/1946) issue, I have to use the latest checkout as of writing.

dpgeorge, 2016-04-04:
The documentation you link to is for the pyboard, not the esp8266.

To run boot.py you need to have the filesystem mounted.  At the REPL, if you do import os, then os.listdir() it should list boot.py and/or main.py.

pfalcon, 2016-04-11:
https://github.com/micropython/micropython/commit/6e87aeb8419e8c5fe009577a3b1bb18254a35b79 implements new bootstrapping sequence.

closed 2016-04-11 by pfalcon