micropython/micropython #1968: cc3200/bootmgr/relocator/relocator.bin

jhdulaney, 2016-04-05:
Hello,

What is the nature of this file?  What license is it under?

Commit is here:

https://github.com/micropython/micropython/commit/f8146d021ba933034e7a6244e4267a6626357cf7

danicampora, 2016-04-05:
> f8146d0

Hi,

It's part of the CC3200 SDK supplied by TI. It is licensed the same way 
as the other files from Texas Instruments. The source code is also 
available in the SDK, I just used the .bin for convenience...

Cheers, Daniel

jhdulaney, 2016-04-05:
Okay, thanks; it came up as I am looking at packaging for Fedora, and distributing binaries is frowned upon.

jhdulaney, 2016-04-05:
I can't seem to find the text of an actual license?  Looking at the boilerplate headers in some of the files, it looks MIT-ish.

jhdulaney, 2016-04-13:
Okay, I just removed the binary as it's not needed in Fedora, and shipping it is against the package guidelines.

closed 2016-04-13 by jhdulaney