micropython/micropython #1972: Build failure - extmod/modure.c

tomsparks, 2016-04-07:
I'm building on FreeBSD-11 with GCC 4.8.5 and gnu make (gmake).

CC ../py/../extmod/modure.c
../py/../extmod/modure.c:95:17: error: conflicting types for 're_exec'
 STATIC mp_obj_t re_exec(bool is_anchored, uint n_args, const mp_obj_t *args) {
                 ^
In file included from ./mpconfigport.h:235:0,
                 from ../py/mpconfig.h:45,
                 from ../py/nlr.h:36,
                 from ../py/../extmod/modure.c:31:
/usr/local/lib/gcc48/gcc/x86_64-portbld-freebsd11.0/4.8.5/include-fixed/unistd.h:557:6: note: previous declaration of 're_exec' was here
 int  re_exec(const char *);
      ^
../py/mkrules.mk:47: recipe for target 'build/py/../extmod/modure.o' failed
gmake: *** [build/py/../extmod/modure.o] Error 1

Not sure what the deal is with this.

vfl68, 2016-04-07:
For which hardware are you trying to build ?

tomsparks, 2016-04-07:
The "unix" version, I was going to go for the ESP8266 version afterwards.

vfl68, 2016-04-07:
As I can read here http://www.unix.com/man-page/freebsd/3/re_exec/
the function re_exec exist in the freebsd unistd.h file ... and the same function is redefined in micropython. This function isn't defined in the linux unistd.h file.

re_exec doesn't seem to be used outside of modure.c, try to rename every re_exec in the file to something like re_exec_mpy and try to build

tomsparks, 2016-04-07:
ok, I did that, tested, works. I'll submit a pull request.

pfalcon, 2016-04-10:
Fixed in master. @tomsparks : A small hint: you could add your commit email to github account (invisible to others) to better associate ones to another.

closed 2016-04-10 by pfalcon