micropython/micropython #1974: ESP8266/  Build stopped working  on OS X ,  issues with lwip

tritip, 2016-04-07:
Built from scratch, made clean, compiled all source, tried to build by  hand, started to patch the code.  There 's some commented out business in the Makefile  pertaining to using the local lwip library over the sdk version;  perhaps this has not been fully resolved yet.

dpgeorge, 2016-04-08:
Did you do `git submodule update --init`?  Please provide the specific error message from the non-working compile.

tritip, 2016-04-08:
Yes, did the git submodule update --init.   Compile bombed out on the LINK build/firmware.elf  numerous undefined references.  Here is a link to a pastebin with the transcript:   http://pastebin.com/DdKpN6MW

Thanks Damien!

yjroot, 2016-04-18:
I had the same problem as you and I got it fixed.
Try to checkout this for your [esp-open-sdk](https://github.com/pfalcon/esp-open-sdk) submodule and rebuild it.
https://github.com/yjroot/esp-open-lwip/commit/e424e026c901d91bce054f908feec5041f30f0d4

pfalcon, 2016-04-28:
This should be fixed. See also http://forum.micropython.org/viewtopic.php?f=16&t=1689 for comments on build tips and tricks.

closed 2016-04-28 by pfalcon