micropython/micropython #1975: can't cross-compile with buildroot-gcc342 [unix branch]

ruiking, 2016-04-08:
make CROSS_COMPILE=mipsel-linux-

> 
> .....
> CC ../py/mpstate.c
> In file included from ../py/mpstate.h:31,
>                  from ../py/mpstate.c:27:
> ../py/mpconfig.h:29: error: parse error before '/' token
> In file included from ../py/mpstate.h:31,
>                  from ../py/mpstate.c:27:
> ../py/mpconfig.h:36:52: warning: character constant too long for its type
> In file included from ../py/mpconfig.h:45,
>                  from ../py/mpstate.h:31,
>                  from ../py/mpstate.c:27:
> ./mpconfigport.h:118: error: missing terminating ' character
> ./mpconfigport.h:195: error: parse error before '/' token
> ./mpconfigport.h:219: error: parse error before '/' token
> ./mpconfigport.h:223: error: parse error before "mp_uint_t"
> ......

dpgeorge, 2016-04-08:
Looks like it's in strict C mode where the compiler doesn't allow/recognise // as a comment.  Run make with V=1 to see exactly what is being called.

ruiking, 2016-04-08:
make CROSS_COMPILE=mipsel-linux- V=1

> python ../py/makeversionhdr.py build/genhdr/mpversion.h
> CC ../py/mpstate.c
> mipsel-linux-gcc -I../lib/netutils -I. -I.. -Ibuild -I../lib/mp-readline -Wall -Werror -Wpointer-arith -Wuninitialized -ansi -std=gnu99 -DUNIX -DMICROPY_USE_READLINE=1 -DMICROPY_PY_TIME=1 -DMICROPY_PY_TERMIOS=1 -DMICROPY_PY_SOCKET=1   -DMICROPY_PY_FFI=1 -Os   -U _FORTIFY_SOURCE -c -MD -o build/py/mpstate.o ../py/mpstate.c
> In file included from ../py/mpstate.h:32,
>                  from ../py/mpstate.c:27:
> ../py/misc.h:181: error: syntax error before "mp_uint_t"
> ../py/misc.h: In function `count_lead_ones':
> ../py/misc.h:183: error: 'for' loop initial declaration used outside C99 mode
> In file included from ../py/nlr.h:63,
>                  from ../py/mpstate.h:33,
>                  from ../py/mpstate.c:27:
> ../py/mpstate.h:199:8: extra tokens at end of #endif directive
> In file included from ../py/obj.h:31,
>                  from ../py/mpstate.h:34,
>                  from ../py/mpstate.c:27:
> ../py/qstr.h: At top level:
> ../py/qstr.h:32: error: parse error before '/' token
> ../py/qstr.h:32: error: missing terminating ' character
> ../py/qstr.h:36: error: missing terminating ' character
> In file included from ../py/obj.h:31,
>                  from ../py/mpstate.h:34,
>                  from ../py/mpstate.c:27:
> ../py/qstr.h:61: error: parse error before '/' token
> In file included from ../py/obj.h:31,
>                  from ../py/mpstate.h:34,
>                  from ../py/mpstate.c:27:
> ../py/qstr.h:77:8: extra tokens at end of #endif directive
> In file included from ../py/obj.h:32,
>                  from ../py/mpstate.h:34,
>                  from ../py/mpstate.c:27:
> ../py/mpprint.h:49: error: parse error before '/' token

ruiking, 2016-04-08:
mipsel-linux-gcc -I../lib/netutils -I. -I.. -Ibuild -I../lib/mp-readline -Wall -Werror -Wpointer-arith -Wuninitialized -ansi  -DUNIX -DMICROPY_USE_READLINE=1 -DMICROPY_PY_TIME=1 -DMICROPY_PY_TERMIOS=1 -DMICROPY_PY_SOCKET=1 -DMICROPY_PY_FFI=1 -Os -U _FORTIFY_SOURCE -c -MD -o build/py/mpstate.o ../py/mpstate.c

> In file included from ../py/mpstate.h:32,
>                  from ../py/mpstate.c:27:
> ../py/misc.h:29: error: parse error before '/' token
> In file included from ../py/mpstate.h:32,
>                  from ../py/mpstate.c:27:
> ../py/misc.h:51: error: parse error before '/' token
> ../py/misc.h:101: error: parse error before '/' token
> ../py/misc.h:118: warning: type defaults to `int' in declaration of `utf8_get_char'
> ../py/misc.h:118: warning: data definition has no type or storage class
> ../py/misc.h:121: error: parse error before "c"
> ../py/misc.h:122: error: parse error before "c"
> ../py/misc.h:123: error: parse error before "c"
> ../py/misc.h:124: error: parse error before "c"
> ../py/misc.h:125: error: parse error before "c"
> ../py/misc.h:126: error: parse error before "c"
> ../py/misc.h:127: error: parse error before "c"
> ../py/misc.h:128: error: parse error before "c"
> ../py/misc.h:129: error: parse error before "unichar_tolower"
> ../py/misc.h:129: error: parse error before "c"

dpgeorge, 2016-04-08:
Please report the exact version of the compiler.

ruiking, 2016-04-08:
Okay, compiler info:

ruiking, 2016-04-08:
Okay, compiler version info:

mipsel-linux-gcc --version

> mipsel-linux-gcc (GCC) 3.4.2
> Copyright (C) 2004 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

for [Ralink rt5350 ](https://wikidevi.com/wiki/Ralink_RT5350)

ruiking, 2016-04-08:
Here is a sample [env setup instruction](http://ftp.comprousa.com/downloadfiles/RT5350%20Development%20Environment%20Setup%20Instructions.pdf) , thanks!

ruiking, 2016-04-08:
It's an c89 issue, remove **-ansi** directive in Makefile fix it. Thanks.

closed 2016-04-08 by ruiking