micropython/micropython #1976: link error, cross-compile with gcc3.4.2 on rt5350 (mipsel) [unix branch]

ruiking, 2016-04-08:
mipsel-linux-gcc -o micropython build/py/mpstate.o build/py/nlrx86.o build/py/nlrx64.o build/py/nlrthumb.o build/py/nlrxtensa.o build/py/nlrsetjmp.o build/py/malloc.o build/py/gc.o build/py/qstr.o build/py/vstr.o build/py/mpprint.o build/py/unicode.o build/py/mpz.o build/py/lexer.o build/py/lexerstr.o build/py/lexerunix.o build/py/parse.o build/py/scope.o build/py/compile.o build/py/emitcommon.o build/py/emitbc.o build/py/asmx64.o build/py/emitnx64.o build/py/asmx86.o build/py/emitnx86.o build/py/asmthumb.o build/py/emitnthumb.o build/py/emitinlinethumb.o build/py/asmarm.o build/py/emitnarm.o build/py/formatfloat.o build/py/parsenumbase.o build/py/parsenum.o build/py/emitglue.o build/py/runtime.o build/py/nativeglue.o build/py/stackctrl.o build/py/argcheck.o build/py/warning.o build/py/map.o build/py/obj.o build/py/objarray.o build/py/objattrtuple.o build/py/objbool.o build/py/objboundmeth.o build/py/objcell.o build/py/objclosure.o build/py/objcomplex.o build/py/objdict.o build/py/objenumerate.o build/py/objexcept.o build/py/objfilter.o build/py/objfloat.o build/py/objfun.o build/py/objgenerator.o build/py/objgetitemiter.o build/py/objint.o build/py/objint_longlong.o build/py/objint_mpz.o build/py/objlist.o build/py/objmap.o build/py/objmodule.o build/py/objobject.o build/py/objpolyiter.o build/py/objproperty.o build/py/objnone.o build/py/objnamedtuple.o build/py/objrange.o build/py/objreversed.o build/py/objset.o build/py/objsingleton.o build/py/objslice.o build/py/objstr.o build/py/objstrunicode.o build/py/objstringio.o build/py/objtuple.o build/py/objtype.o build/py/objzip.o build/py/opmethods.o build/py/sequence.o build/py/stream.o build/py/binary.o build/py/builtinimport.o build/py/builtinevex.o build/py/modarray.o build/py/modbuiltins.o build/py/modcollections.o build/py/modgc.o build/py/modio.o build/py/modmath.o build/py/modcmath.o build/py/modmicropython.o build/py/modstruct.o build/py/modsys.o build/py/vm.o build/py/bc.o build/py/showbc.o build/py/repl.o build/py/smallint.o build/py/frozenmod.o build/py/../extmod/moductypes.o build/py/../extmod/modujson.o build/py/../extmod/modure.o build/py/../extmod/moduzlib.o build/py/../extmod/moduheapq.o build/py/../extmod/moduhashlib.o build/py/../extmod/modubinascii.o build/py/../extmod/machine_mem.o build/py/../extmod/modussl.o build/py/../extmod/modurandom.o build/py/../extmod/modwebsocket.o build/py/../extmod/fsusermount.o build/py/../extmod/vfs_fat.o build/py/../extmod/vfs_fat_ffconf.o build/py/../extmod/vfs_fat_diskio.o build/py/../extmod/vfs_fat_file.o build/py/../extmod/vfs_fat_lexer.o build/py/../extmod/vfs_fat_misc.o build/py/../extmod/moduos_dupterm.o build/main.o build/gccollect.o build/unix_mphal.o build/input.o build/file.o build/modmachine.o build/modos.o build/moduselect.o build/alloc.o build/coverage.o build/fatfs_port.o build/modtime.o build/modtermios.o build/modsocket.o build/modffi.o build/build/_frozen_upip.o build/lib/mp-readline/readline.o build/lib/utils/printf.o build/lib/fatfs/ff.o build/lib/fatfs/option/ccsbcs.o  ../lib/libffi/build_dir/out/lib/libffi.a -lm -static-libgcc -ldl -Wl,-Map=micropython.map,--cref

> build/py/modbuiltins.o: In function `mp_builtin_round':modbuiltins.c:(.text+0x1124): undefined reference to `round'
> :modbuiltins.c:(.text+0x1160): undefined reference to `round'
> build/py/modmath.o: In function `mp_math_trunc':modmath.c:(.text+0xb5c): undefined reference to `trunc'
> build/py/modmath.o: In function `mp_math_gamma':modmath.c:(.text+0xd38): undefined reference to `tgamma'
> build/modtime.o: In function `mod_time_sleep':modtime.c:(.text+0x258): undefined reference to `round'

ruiking, 2016-04-10:
Looks like it's an [old gcc issue](http://stackoverflow.com/questions/575936/how-am-i-incorrectly-using-the-round-function-in-c), any suggestions?

pfalcon, 2016-06-18:
MicroPython requires gcc with solid C99 support, i.e. at least the latest of 4.x series (4.8, etc.)

closed 2016-06-18 by pfalcon