micropython/micropython #1977: build system: make clean does not remove build files

pohmelie, 2016-04-09:
`make clean` trying to remove only
```
rm -f micropython
rm -f micropython.map
rm -rf build
```
But `freedos` and `nanbox` targets build directory and binary name differs. Even they have `BUILD` and `PROG` values in `Makefile`.

pfalcon, 2016-04-09:
> Even they have BUILD and PROG values in Makefile.

Then run it correspondingly and it will work as you expect:

~~~
make clean BUILD=build-nanbox PROG=micropython_nanbox
~~~

pohmelie, 2016-04-09:
Hm… Is this a common way to clean different targets? It looks verbose a little. Maybe something like `make clean nanbox`? Or it's impossible?

pfalcon, 2016-04-09:
The common way to clean anything is rm -rf or few clicks/keypresses in your favorite filemanager ;-). Makefile can't do all possible things (not without growing too big and unwieldy).

`make clean nanbox` won't work (well, maybe if using too much dirty make magic). More realistic `make clean-nanbox`, but that means more verbosity in Makefile, again.

@dhylands : Do you have any comments?

dhylands, 2016-04-09:
One quite simple approach would be to just have clean do the clean for all of the targets (minimal changes required).

The other approach is to do something like what what stmhal does, but perhaps use a different variable. This would mean that instead of doing:
```
make freedos
```
you'd do something like
```
make APP=freedos
```
then to clean you would do:
```
make APP=freedos clean
```
and if you were working on one particular target alot, then you could set the APP env var or setup a GNUmakefile which sets APP and inludes the regular Makefile. The variable chosen (APP in my example) is arbitrary. I'd probably choose BUILD, but then that might cause confusion with the current scheme.

Personally, I've always preferred the latter approach (using a variable like APP=) as it makes the makefile cleaner. The current solution has make calling itself recursively, which I definitely have a distaste for (just my personal preference).

Having to specify both BUILD and PROG is really just a hack due to the history of the way the current makefile evolved.

dhylands, 2016-04-09:
> Hm… Is this a common way to clean different targets? It looks verbose a little. Maybe something like make clean nanbox? Or it's impossible?

It is possible to do this by some makefile magic. make creates a variable MAKECMDGOALS and you can detect that freedos is listed in there. But it introduces a bunch of other complexities (which I can elaborate on if desired).

I prefer to be a bit more explicit by having the variable assigned on the command line.

pohmelie, 2016-04-09:
> just have clean do the clean for all of the targets

Vote for this.

dpgeorge, 2016-04-09:
I would vote to change to the `make APP=freedos` approach.  Apart from fixing the clean issue (which is not really a big issue IMO) it would allow you to run tests (eg make APP=fast test), install/uninstall, etc, which is something that I've been missing for a while now.

dpgeorge, 2022-06-23:
Unix now uses `make VARIANT=...`.

closed 2022-06-23 by dpgeorge