micropython/micropython #1978: esp8266 make error

Adixylian, 2016-04-09:
Greetings

I have successfully compiled few of the esp8266 versions with esp-open-sdk and flashed them with esptool
But now I cant get it to work, after I updated esp-open-sdk

This error shows every time:

GEN build/genhdr/qstrdefs.generated.h
xtensa-lx106-elf-gcc: error: unrecognized command line option '-mforce-l32'

it's impossible for me to fix it. I'm not a Linux guy and it's new to me so I can't do it by myself. Actually, I started using Linux when I bought the esp8266, few weeks ago...

pfalcon, 2016-04-09:
This flag is in esp-open-sdk for more than month. Please ask for newbie help at http://forum.micropython.org/viewforum.php?f=16 .

closed 2016-04-09 by pfalcon

bmsleight, 2016-10-01:
Using the instructions as written resulted in this behaviour for myself.

Solution: Install xtensa-lx106-elf via https://github.com/pfalcon/esp-open-sdk
Then
 export PATH=/home/bms/micropython/esp-open-sdk/xtensa-lx106-elf/bin:$PATH

Hope this helps others